FreeOpcUa/python-opcua #1109: Getting BadIdentityTokenInvalid error on attempting initial connection to Kaiser Raman server...

sb934, 2020-07-22:
----------
    _C:\Users\YD\Desktop\kaiser-opcua (1)\kaiser-opcua>C:/Users/YD/AppData/Local/Programs/Python/Python38-32/python.exe "c:/Users/YD/Desktop/kaiser-opcua (1)/kaiser-opcua/client.py"
WARNING:opcua.crypto.security_policies:DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received  in response to ActivateSessionRequest
Traceback (most recent call last):
  File "c:/Users/YD/Desktop/kaiser-opcua (1)/kaiser-opcua/client.py", line 22, in <module>
    client.connect()
  File "C:\Users\YD\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\client\client.py", line 279, in connect
    self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)
  File "C:\Users\YD\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\client\client.py", line 464, in activate_session
    return self.uaclient.activate_session(params)
  File "C:\Users\YD\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\client\ua_client.py", line 299, in activate_session
    data = self._uasocket.send_request(request)
  File "C:\Users\YD\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\client\ua_client.py", line 84, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "C:\Users\YD\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\YD\AppData\Local\Programs\Python\Python38-32\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadIdentityTokenInvalid: "The user identity token is not valid."(BadIdentityTokenInvalid)_

sb934, 2020-07-23:
I got it. The server supported username, password-based authentication. I was passing additional certificates and private keys that didn't need to be there. Thank you though @AndreasHeine. You've been a great help. I'll keep tagging you in my other OPC UA issues 🦺

closed 2020-07-23 by sb934

mirkocomparetti-synesis, 2021-04-15:
@sb934 can you share the "ActivateSession" line?
I'm facing the same issue and I have this line
```self.clientOPCUA.activate_session(username=self.opcua_user, password=self.opcua_passw)```
Thanks!