FreeOpcUa/python-opcua #1111: read_raw_history return no error when no data is available

k-sartor, 2020-07-23:
I try to read some historical values with read_raw_history.
All works except when the initial date is over the date available on the server. In this case, no error is shown and i don't know how treat this case.
Here the code
    ```
    client.connect()
        objects = client.get_objects_node()
        struct = client.get_node("ns=1;s=t|B37_Math_DB_Sondes_AI_T_Amb1_Stat_Sud")
        counter1 = struct 
        starttime = "01.05.2020 10:00:00"  
        starttime_dt_format = datetime.datetime.strptime(starttime,"%d.%m.%Y %H:%M:%S")  
        endtime = "23.07.2020 10:00:00"  
        endtime_dt_format = datetime.datetime.strptime(endtime,"%d.%m.%Y %H:%M:%S")  
        try:
            evs=counter1.read_raw_history(starttime=starttime_dt_format, endtime=endtime_dt_format, numvalues=10000)  
            for ev in evs:
                print(ev)
        except Exception as e:
            print("Erreur"+str(e))
```
In this case e variable is empty, no data are returned in evs variable so i can't treat this case...

zerox1212, 2020-07-23:
It would be helpful to know what server you are connected to.

k-sartor, 2020-07-24:
To be honest, i know only it is WinCC Server with a opcua licence => I use this current librairy to get the data 
Is there any function to give you the wished information?