FreeOpcUa/python-opcua #1113: what best way to subscribe nodes which are dynamically changing?

SeongsuByeon, 2020-08-05:
I want to subscribe several nodes with browsing path. 
As you can see below, there are many identification numbers in the AIS node.
And each identification number has several variables inside.

It is challenging to come up with idea to read all these nodes in the loop because the node name (Ex M03023045) is keep changing. when there was one node with static node name, it was simple to read all variables of this node with subscription.
But in this case, what is the best approach to read all variables?
 
[image]
[image]

swamper123, 2020-08-05:
So does the Browsername change and the values inside stay the same? 
Or do you create new Nodeobjects in File `Class_A` and want to add them to your subscription list?

SeongsuByeon, 2020-08-05:
@swamper123 I want to get values from client side. And these identification numbers(ex,M2345232) are keep changing with its values. I figured it out to get the values from each node as below. My plan is to create several subscriptions for all identification numbers. But problem is that I am getting error. I attached detailed information. 

`class TrafficShipData(object):
    
    def __init__(self, opcua_server,nodes):
        self.ua_node = nodes
        self.Cog_browsename = nodes[0].get_browse_name().Name
        self.Heading_browsename = nodes[1].get_browse_name().Name

        self.CourseOverGround_OS = 0
        self.Heading_OS = 0
        self.Latitude_OS = 0
        self.Longitude_OS = 0
        self.RateOfTurn_OS = 0
        self.SpeedOverGround_OS = 0
        self.SpeedThroughWater_OS = 0
    
        handler = SubHandler_trafficShip(self)
        sub = opcua_server.create_subscription(1000, handler)
        handle = sub.subscribe_data_change(self.ua_node)
`
`        ClassAObjList = []
        AIS_node_ClassA = client.nodes.objects.get_child(['{}:IMO9796042'.format(idx), '{}:NavigationalData'.format(idx), '{}:AIS'.format(idx),'{}:Class_A'.format(idx)])
        AIS_node_ClassA_list = AIS_node_ClassA.get_children()
        for i in len(AIS_node_ClassA.get_children()):
            
            CourseOverGround_TS = AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Course'.format(idx)])
            Heading_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Heading'.format(idx)])
            SpeedOverGround_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Speed'.format(idx)])
            Latitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Latitude'.format(idx)])
            Longitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Longitude'.format(idx)])
            RateOfTurn_TS =  AIS_node_ClassA_list[i].get_child(['{}:RateOfTurn'.format(idx)])
            
            point_a_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_a'.format(idx)])
            point_b_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_b'.format(idx)])
            point_c_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_c'.format(idx)])
            point_d_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_d'.format(idx)])
            
            ts_nodes = [CourseOverGround_TS,Heading_TS,SpeedOverGround_TS,Latitude_TS,Longitude_TS,RateOfTurn_TS,point_a_TS,point_b_TS,point_c_TS,point_d_TS]
            TSobject_ + str(1) = TrafficShipData (client,ts_nodes) 
            ClassAObjList.append(TSobject)`

> ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x28291196c48 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: "The session was closed by the client."(BadSessionClosed)
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x2829119fc88 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: "The session was closed by the client."(BadSessionClosed)

SeongsuByeon, 2020-08-05:
@swamper123 I want to get values from client side. And these identification numbers(ex,M2345232) are keep changing with its values. I figured it out to get the values from each node as below. My plan is to create several subscriptions for all identification numbers. But problem is that I am getting error. I attached detailed information. 

ClassAObjList = []
AIS_node_ClassA = client.nodes.objects.get_child(['{}:IMO9796042'.format(idx), '{}:NavigationalData'.format(idx), '{}:AIS'.format(idx),'{}:Class_A'.format(idx)])
AIS_node_ClassA_list = AIS_node_ClassA.get_children()
for i in len(AIS_node_ClassA.get_children()):
    
    CourseOverGround_TS = AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Course'.format(idx)])
    Heading_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Heading'.format(idx)])
    SpeedOverGround_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Speed'.format(idx)])
    Latitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Latitude'.format(idx)])
    Longitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Longitude'.format(idx)])
    RateOfTurn_TS =  AIS_node_ClassA_list[i].get_child(['{}:RateOfTurn'.format(idx)])
    
    point_a_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_a'.format(idx)])
    point_b_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_b'.format(idx)])
    point_c_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_c'.format(idx)])
    point_d_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_d'.format(idx)])
    
    ts_nodes = [CourseOverGround_TS,Heading_TS,SpeedOverGround_TS,Latitude_TS,Longitude_TS,RateOfTurn_TS,point_a_TS,point_b_TS,point_c_TS,point_d_TS]
    TSobject_ + str(1) = TrafficShipData (client,ts_nodes) 
    `ClassAObjList.append(TSobject)`

> ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x28291196c48 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: "The session was closed by the client."(BadSessionClosed)
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x2829119fc88 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: "The session was closed by the client."(BadSessionClosed)

swamper123, 2020-08-05:
But the NodeID be the same after changing?

SeongsuByeon, 2020-08-05:
@swamper123 node id is not changing. it is changed when we reboot the server. 

`ClassAObjList = []
AIS_node_ClassA = client.nodes.objects.get_child(['{}:IMO9796042'.format(idx), '{}:NavigationalData'.format(idx), '{}:AIS'.format(idx),'{}:Class_A'.format(idx)])
AIS_node_ClassA_list = AIS_node_ClassA.get_children()
for i in len(AIS_node_ClassA.get_children()):
    
    CourseOverGround_TS = AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Course'.format(idx)])
    Heading_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Heading'.format(idx)])
    SpeedOverGround_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Speed'.format(idx)])
    Latitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Latitude'.format(idx)])
    Longitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Longitude'.format(idx)])
    RateOfTurn_TS =  AIS_node_ClassA_list[i].get_child(['{}:RateOfTurn'.format(idx)])
    
    point_a_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_a'.format(idx)])
    point_b_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_b'.format(idx)])
    point_c_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_c'.format(idx)])
    point_d_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_d'.format(idx)])
    
    ts_nodes = [CourseOverGround_TS,Heading_TS,SpeedOverGround_TS,Latitude_TS,Longitude_TS,RateOfTurn_TS,point_a_TS,point_b_TS,point_c_TS,point_d_TS]
    TSobject_ + str(1) = TrafficShipData (client,ts_nodes) 
    ClassAObjList.append(TSobject)`

SeongsuByeon, 2020-08-05:
``ClassAObjList = []
AIS_node_ClassA = client.nodes.objects.get_child(['{}:IMO9796042'.format(idx), '{}:NavigationalData'.format(idx), '{}:AIS'.format(idx),'{}:Class_A'.format(idx)])
AIS_node_ClassA_list = AIS_node_ClassA.get_children()
for i in len(AIS_node_ClassA.get_children()):
    
    CourseOverGround_TS = AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Course'.format(idx)])
    Heading_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Heading'.format(idx)])
    SpeedOverGround_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Speed'.format(idx)])
    Latitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Latitude'.format(idx)])
    Longitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Longitude'.format(idx)])
    RateOfTurn_TS =  AIS_node_ClassA_list[i].get_child(['{}:RateOfTurn'.format(idx)])
    
    point_a_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_a'.format(idx)])
    point_b_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_b'.format(idx)])
    point_c_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_c'.format(idx)])
    point_d_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_d'.format(idx)])
    
    ts_nodes = [CourseOverGround_TS,Heading_TS,SpeedOverGround_TS,Latitude_TS,Longitude_TS,RateOfTurn_TS,point_a_TS,point_b_TS,point_c_TS,point_d_TS]
    TSobject_ + str(1) = TrafficShipData (client,ts_nodes) 
    ClassAObjList.append(TSobject)``

SeongsuByeon, 2020-08-05:
`123`

SeongsuByeon, 2020-08-05:
@swamper123 node id is constant. It is changed only after rebooting of the server. 
But why am I getting error as above? if I set a time sleep after "handle = sub.subscribe_data_change(self.ua_node", I do not get error. 
[image]
[image]

SeongsuByeon, 2020-08-05:
@swamper123  node ID is constant. 
But my question is that 

`class TrafficShipData(object):
    
    def __init__(self, opcua_server,nodes):
        self.ua_node = nodes
        self.Cog_browsename = nodes[0].get_browse_name().Name
        self.Heading_browsename = nodes[1].get_browse_name().Name

        self.CourseOverGround_OS = 0
        self.Heading_OS = 0
        self.Latitude_OS = 0
        self.Longitude_OS = 0
        self.RateOfTurn_OS = 0
        self.SpeedOverGround_OS = 0
        self.SpeedThroughWater_OS = 0
    
        handler = SubHandler_trafficShip(self)
        sub = opcua_server.create_subscription(1000, handler)
        handle = sub.subscribe_data_change(self.ua_node)
# =============================================================================
#         time.sleep(100)
#         sub.unsubscribe(handle)
#         sub.delete()    
# ===============================`

SeongsuByeon, 2020-08-05:
@swamper123  both node id and name are changing. 
what is the best way to create multiple subscriptions? it looks like I have to use multi-threading.

swamper123, 2020-08-06:
I haven't tried it yet with python-opcua, but perhaps you can add an datachange_notification

swamper123, 2020-08-06:
Would you may add your solution, in case this request spawns again?

I am not so familiar with` python-opcua`, more with `opcua-asyncio` ( I would prefere in general, since this repo is in maintainance mode), but it seems to be like you say, that every time a thread is spawned. I wouldn't know a good alternative at the moment, maybe somebody else has a good idea.

While crawling through the deeps of the specs, I found [DataSet](https://reference.opcfoundation.org/v104/Core/docs/Part14/5.2.1/) and `DataSetFolder`, which may be should mentioned here for future purposes.

SeongsuByeon, 2020-08-06:
@swamper123 I added this embed() function after starting subscription. I found this example from this repository. 

`    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()`

zerox1212, 2020-08-06:
If you search through the issues for multithreading there are many problems. I would avoid that approach if at all possible because it can cause very weird behavior.

Why do you need multi threading, you can make as many subscription as you want without doing that.

SeongsuByeon, 2020-08-06:
@zerox1212  thank you for your advice. 
my plan is to make subscriptions for all different traffic ship's nodes and create objects for each ships.
Because I have to complete one XML documents. 

So the problem is that subscription for one node is blocking other subscriptions when I use the same class. 
I implemented multithreading for this reason. 

`class TrafficShipData(object):
    
    def __init__(self,opcua_server,nodes):
        self.ua_node = nodes
        self.CourseOverGround_TS = 0
        self.Heading_TS = 0
        self.SpeedOverGround_TS = 0
        self.Latitude_TS = 0
        self.Longitude_TS = 0
        self.RateOfTurn_TS = 0
        self.point_a_TS = 0
        self.point_b_TS = 0
        self.point_c_TS = 0
        self.point_d_TS = 0
        self.client = opcua_server
        
        handler = SubHandler_trafficShip(self)
        sub = opcua_server.create_subscription(1000, handler)
        handle = sub.subscribe_data_change(self.ua_node)
        while True: 
         time.sleep(100)
         break
        sub.unsubscribe(handle)
        sub.delete()       

# =============================================================================
#     def subscription(self):
#         print("traffic subscription started")
#         handler = SubHandler_trafficShip(self)
#         sub = self.client.create_subscription(1000, handler)
#         handle = sub.subscribe_data_change(self.ua_node)
#         while True: 
#          time.sleep(100)
#          break
#         sub.unsubscribe(handle)
#         sub.delete()         
# =============================================================================
    
    def set_CourseOverGround_TS(self,course):
        self.CourseOverGround_TS = course
   
    def set_Heading_TS(self,Heading):
        self.Heading_TS = Heading
        
    def set_SpeedOverGround_TS(self,SpeedOverGround):
        self.SpeedOverGround_TS =  SpeedOverGround   `

SeongsuByeon, 2020-08-06:
@zerox1212 

[image]
[image]

zerox1212, 2020-08-07:
I still don't get it or I don't understand what your server is doing.

Why can't you just browse Class_A folder every few seconds, and if you find a new Object, create a subscription to all it's child nodes? When you create a new subscription, keep track of it in a `dict` with the parents `browsename` as the key or something similar.

SeongsuByeon, 2020-08-10:
@zerox1212 Thank you for your explanation! 
First of all, our server is monitoring all traffic ships information. For instance, you have navigational equipment which can detect ships around your own ship. And the the detection is based on the distance between traffic ships and my own ship. So it is important to subscribe each traffic ships dynamically. There are several types of traffic ships like Class A and Class B.

**I got an information about the server from our partner as below.** 
"As the number of AIS targets is changing dynamically, we need to monitor add / remove node events on the server. The client API should provide methods to handle such events. 
In example with the UA C++ SDK, which we are using, you would need to reimplement the newEvents method. Which client SDK are you using?
When receiving an add node event you can subscribe to this node's variables as usual (datachange_notification). On remove node you will remove the respective subscription. 

@zerox1212 I tried that approach. But creating subscription to all it's child nodes was problematic. 
`        AIS_node_ClassA = client.nodes.objects.get_child(['{}:IMO9796042'.format(idx), '{}:NavigationalData'.format(idx), '{}:AIS'.format(idx),'{}:Class_A'.format(idx)])
        AIS_node_ClassA_list = AIS_node_ClassA.get_children()
        for i in range(5):# len(AIS_node_ClassA.get_children()):
            
            CourseOverGround_TS = AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Course'.format(idx)])
            Heading_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Heading'.format(idx)])
            SpeedOverGround_TS =  AIS_node_ClassA_list[i].get_child(['{}:AttitudeData'.format(idx),'{}:Speed'.format(idx)])
            Latitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Latitude'.format(idx)])
            Longitude_TS =  AIS_node_ClassA_list[i].get_child(['{}:Position'.format(idx),'{}:Longitude'.format(idx)])
            RateOfTurn_TS =  AIS_node_ClassA_list[i].get_child(['{}:RateOfTurn'.format(idx)])
            
            point_a_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_a'.format(idx)])
            point_b_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_b'.format(idx)])
            point_c_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_c'.format(idx)])
            point_d_TS =  AIS_node_ClassA_list[i].get_child(['{}:ShipsDimension'.format(idx),'{}:rpoint_d'.format(idx)])
            
    
            ts_nodes = [CourseOverGround_TS,Heading_TS,SpeedOverGround_TS,Latitude_TS,Longitude_TS,RateOfTurn_TS,point_a_TS,point_b_TS,point_c_TS,point_d_TS]
            t = threading.Thread(target=worker, args=(client,ts_nodes))`

AndreasHeine, 2020-08-10:
you need to read cyclic the children and then you have to subscribe all children
an on next cycle you need to compare subscribed nodes against new children (newer get_children) an if its not in subscribet node-list append and subscribe if its not in newer (means out of range) you can unsub this node handle

hint:
```python
def Diff(li1, li2): 
	return (list(set(li1) - set(li2))) 

li1 = [10, 15, 20, 25, 30, 35, 40] 
li2 = [25, 40, 35] 
print(Diff(li1, li2)) 
```

C:\Users\andre>python c:/Users/andre/Desktop/test.py
[10, 20, 30, 15]

its a little more complex then that but it is just structured list handling so its possible if you make the structure right!

zerox1212, 2020-08-11:
@AndreasHeine That is not really what he needs to do. According to his post above the server is generating an event whenever it adds/removes a ship. His client needs to subscribe to the server events and when the client receives and update he should immediately subscribe/unsubscribe from the new/removed node.

His problem is that this libraries client isn't async. I'm really only familiar with the server so I don't know the clients limitations. Can this client really only create a single subscription or is he doing something wrong?

zerox1212, 2020-08-11:
@SeongsuByeon can you post the example using UA C++ SDK that is mentioned? With this example we can just reproduce what they did but in Python.

AndreasHeine, 2020-08-11:
@zerox1212 if there are limitations they are serverside like max monitored items (some servers have these limits)... but for clients or in general its better to use asyncua over python-opcua!

SeongsuByeon, 2020-08-11:
@AndreasHeine  Thank you for your answer. Why asyncua is better than pyhton-opcua?

SeongsuByeon, 2020-08-11:
@zerox1212 I will post it as soon as I get the example. 
I have one question about multiple subscription. As I increase the number of subscription over 5, I am getting this error as below.

> ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x1f577eb0dc8 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\sbyeon\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadTooManyPublishRequests: "The server has reached the maximum number of queued publish requests."(BadTooManyPublishRequests)
ServiceFault from server received while waiting for publish response

But my code is keep running and getting updated values from all subscription. 
From sever side, there is no limitation of monitored items and subscription.
Have you ever seen this kind of problem before?

Thank you for your time

SeongsuByeon, 2020-08-11:
@zerox1212  This is the example code from our partner in qt c++

`void ISUAClient::ISUAClientPrivate::newEvents(OpcUa_UInt32 clientSubscriptionHandle, UaEventFieldLists &eventFieldList)
{
    if( clientSubscriptionHandle==0) // modelChangeSubscription
    {
        for ( OpcUa_UInt32 i=0; i<eventFieldList.length(); i++ )
        {
            if(eventFieldList[i].NoOfEventFields<2)
                LOG_ERROR("Invalid event from " + QString::number(clientSubscriptionHandle));
            else
            {
                // First event field is the Message
                UaVariant message    = eventFieldList[i].EventFields[0];
                QString msg = UAConvert::UaToQString(message.toString());
//                LOG_DEBUG("Subscription " + QString::number(clientSubscriptionHandle) + ": Event received - " + msg);

                // Second event field is a list of changes
                // Check type of event field - we expect an extension object array
                OpcUa_Variant *pValue = &eventFieldList[i].EventFields[1];

                if (pValue->Datatype == OpcUaType_ExtensionObject &&
                    pValue->ArrayType == OpcUa_VariantArrayType_Array &&
                    pValue->Value.Array.Length != 0 &&
                    pValue->Value.Array.Value.ExtensionObjectArray != OpcUa_Null)
                {
                    OpcUa_ModelChangeStructureDataType *pNodeChange = getModelChangeObject(&pValue->Value.Array,0);

                    if(pNodeChange!=NULL)
                    {
                        int x = msg.indexOf(": "); // Message contains new node's browse name ("Added node: ..")
                        QString sNew = msg.mid(x>0?x+2:0);
                        x = sNew.lastIndexOf("/");
                        QString sNewName = (x>0)?sNew.mid(x+1):sNew;
                        UaNodeId nid = UaNodeId(pNodeChange->Affected);
                        IsUaNodeId nidAffected(QString("%1").arg(sNew), nid, pNodeChange->AffectedType.Identifier.Numeric);

                        if((pNodeChange->Verb & OpcUa_ModelChangeStructureVerbMask_NodeAdded)==OpcUa_ModelChangeStructureVerbMask_NodeAdded)
                        {
                            OpcUa_ModelChangeStructureDataType *pParentNodeChange = getModelChangeObject(&pValue->Value.Array,1);
                            QString sParent;
                            IsUaNodeId nidParent;
                            if(pParentNodeChange!=NULL)
                            {
                                sParent = UAConvert::UaNodeIdToQString(pParentNodeChange->Affected);
                                nid = UaNodeId(pParentNodeChange->Affected);
                                nidParent = IsUaNodeId((x > 0) ? sNew.left(x)  :"/", nid);
                            }

//                            LOG_DEBUG("New node: " + sNew);
                            emit _self.onModelAddNode(nidParent, nidAffected, sNewName);
                        }
                        else if((pNodeChange->Verb & OpcUa_ModelChangeStructureVerbMask_NodeDeleted)==OpcUa_ModelChangeStructureVerbMask_NodeDeleted)
                        {
//                            LOG_DEBUG("Removed node: " + sNew);
                            emit _self.onModelRemoveNode(nidAffected, sNewName);
                        }
                    }
                }
            }
        }// for
    }
}
`

SeongsuByeon, 2020-08-11:
@zerox1212 I got an example code in qt c++ from our partner as below.

void ISUAClient::ISUAClientPrivate::newEvents(OpcUa_UInt32 clientSubscriptionHandle, UaEventFieldLists &eventFieldList)
{
    if( clientSubscriptionHandle==0) // modelChangeSubscription
    {
        for ( OpcUa_UInt32 i=0; i<eventFieldList.length(); i++ )
        {
            if(eventFieldList[i].NoOfEventFields<2)
                LOG_ERROR("Invalid event from " + QString::number(clientSubscriptionHandle));
            else
            {
                // First event field is the Message
                UaVariant message    = eventFieldList[i].EventFields[0];
                QString msg = UAConvert::UaToQString(message.toString());
//                LOG_DEBUG("Subscription " + QString::number(clientSubscriptionHandle) + ": Event received - " + msg);

                // Second event field is a list of changes
                // Check type of event field - we expect an extension object array
                OpcUa_Variant *pValue = &eventFieldList[i].EventFields[1];

                if (pValue->Datatype == OpcUaType_ExtensionObject &&
                    pValue->ArrayType == OpcUa_VariantArrayType_Array &&
                    pValue->Value.Array.Length != 0 &&
                    pValue->Value.Array.Value.ExtensionObjectArray != OpcUa_Null)
                {
                    OpcUa_ModelChangeStructureDataType *pNodeChange = getModelChangeObject(&pValue->Value.Array,0);

                    if(pNodeChange!=NULL)
                    {
                        int x = msg.indexOf(": "); // Message contains new node's browse name ("Added node: ..")
                        QString sNew = msg.mid(x>0?x+2:0);
                        x = sNew.lastIndexOf("/");
                        QString sNewName = (x>0)?sNew.mid(x+1):sNew;
                        UaNodeId nid = UaNodeId(pNodeChange->Affected);
                        IsUaNodeId nidAffected(QString("%1").arg(sNew), nid, pNodeChange->AffectedType.Identifier.Numeric);

                        if((pNodeChange->Verb & OpcUa_ModelChangeStructureVerbMask_NodeAdded)==OpcUa_ModelChangeStructureVerbMask_NodeAdded)
                        {
                            OpcUa_ModelChangeStructureDataType *pParentNodeChange = getModelChangeObject(&pValue->Value.Array,1);
                            QString sParent;
                            IsUaNodeId nidParent;
                            if(pParentNodeChange!=NULL)
                            {
                                sParent = UAConvert::UaNodeIdToQString(pParentNodeChange->Affected);
                                nid = UaNodeId(pParentNodeChange->Affected);
                                nidParent = IsUaNodeId((x > 0) ? sNew.left(x)  :"/", nid);
                            }

//                            LOG_DEBUG("New node: " + sNew);
                            emit _self.onModelAddNode(nidParent, nidAffected, sNewName);
                        }
                        else if((pNodeChange->Verb & OpcUa_ModelChangeStructureVerbMask_NodeDeleted)==OpcUa_ModelChangeStructureVerbMask_NodeDeleted)
                        {
//                            LOG_DEBUG("Removed node: " + sNew);
                            emit _self.onModelRemoveNode(nidAffected, sNewName);
                        }
                    }
                }
            }
        }// for
    }
}

SeongsuByeon, 2020-08-11:
`void ISUAClient::ISUAClientPrivate::newEvents(OpcUa_UInt32 clientSubscriptionHandle, UaEventFieldLists &eventFieldList)
{
    if( clientSubscriptionHandle==0) // modelChangeSubscription
    {
        for ( OpcUa_UInt32 i=0; i<eventFieldList.length(); i++ )
        {
            if(eventFieldList[i].NoOfEventFields<2)
                LOG_ERROR("Invalid event from " + QString::number(clientSubscriptionHandle));
            else
            {
                // First event field is the Message
                UaVariant message    = eventFieldList[i].EventFields[0];
                QString msg = UAConvert::UaToQString(message.toString());
//                LOG_DEBUG("Subscription " + QString::number(clientSubscriptionHandle) + ": Event received - " + msg);

                // Second event field is a list of changes
                // Check type of event field - we expect an extension object array
                OpcUa_Variant *pValue = &eventFieldList[i].EventFields[1];

                if (pValue->Datatype == OpcUaType_ExtensionObject &&
                    pValue->ArrayType == OpcUa_VariantArrayType_Array &&
                    pValue->Value.Array.Length != 0 &&
                    pValue->Value.Array.Value.ExtensionObjectArray != OpcUa_Null)
                {
                    OpcUa_ModelChangeStructureDataType *pNodeChange = getModelChangeObject(&pValue->Value.Array,0);

                    if(pNodeChange!=NULL)
                    {
                        int x = msg.indexOf(": "); // Message contains new node's browse name ("Added node: ..")
                        QString sNew = msg.mid(x>0?x+2:0);
                        x = sNew.lastIndexOf("/");
                        QString sNewName = (x>0)?sNew.mid(x+1):sNew;
                        UaNodeId nid = UaNodeId(pNodeChange->Affected);
                        IsUaNodeId nidAffected(QString("%1").arg(sNew), nid, pNodeChange->AffectedType.Identifier.Numeric);

                        if((pNodeChange->Verb & OpcUa_ModelChangeStructureVerbMask_NodeAdded)==OpcUa_ModelChangeStructureVerbMask_NodeAdded)
                        {
                            OpcUa_ModelChangeStructureDataType *pParentNodeChange = getModelChangeObject(&pValue->Value.Array,1);
                            QString sParent;
                            IsUaNodeId nidParent;
                            if(pParentNodeChange!=NULL)
                            {
                                sParent = UAConvert::UaNodeIdToQString(pParentNodeChange->Affected);
                                nid = UaNodeId(pParentNodeChange->Affected);
                                nidParent = IsUaNodeId((x > 0) ? sNew.left(x)  :"/", nid);
                            }

//                            LOG_DEBUG("New node: " + sNew);
                            emit _self.onModelAddNode(nidParent, nidAffected, sNewName);
                        }
                        else if((pNodeChange->Verb & OpcUa_ModelChangeStructureVerbMask_NodeDeleted)==OpcUa_ModelChangeStructureVerbMask_NodeDeleted)
                        {
//                            LOG_DEBUG("Removed node: " + sNew);
                            emit _self.onModelRemoveNode(nidAffected, sNewName);
                        }
                    }
                }
            }
        }// for
    }
}
`

SeongsuByeon, 2020-08-11:
@zerox1212 @AndreasHeine 
I am getting event notification from the server as below. 
I know that there is a "event filter" to filter out specific events. Is there any simple example or hint in python opcua? 
In my case, first step has to be catching the relevant event from "event_notification".  But there is no documentation about the opcua.common.event module (https://python-opcua.readthedocs.io/en/latest/opcua.common.html).  

Event when the node is removed from the server

> Traffiship: New event Event(["EventId:b'\\xf6\\xb1$\\x1e\\n,I\\x98\\x97e\\xcf\\xb7~q\\x1b\\xef'", 'EventType:FourByteNodeId(i=2133)', 'SourceNode:FourByteNodeId(i=2253)', 'SourceName:Server', 'Time:2020-08-11 12:43:05.499085', 'ReceiveTime:2020-08-11 12:43:05.499086', 'LocalTime:None', 'Message:LocalizedText(Encoding:3, Locale:, Text**:Removed node: D0022**)', 'Severity:200'])

Event when the node is added 

`Traffiship: New event Event(["EventId:b'\\xcc\\x88J\\x9f\\x16\\xc4D\\x00\\xb3[7\\xa6\\x13\\x9a\\xae\\x84'", 'EventType:FourByteNodeId(i=2133)', 'SourceNode:FourByteNodeId(i=2253)', 'SourceName:Server', 'Time:2020-08-11 13:48:41.276336', 'ReceiveTime:2020-08-11 13:48:41.276337', 'LocalTime:None', 'Message:LocalizedText(Encoding:3, Locale:, Text:**Added node: IMO9796042/NavigationalData/RoutesAndWaypoints/Waypoints/D0023**)', 'Severity:200'])`

AndreasHeine, 2020-08-11:
async is better if you make network requests like http or opcua because you dont block the program (non blocking) you can make multiple requests without waiting till request is finished there some good videos on youtube about python asyncio!

zerox1212, 2020-08-11:
OPC UA Event Filter and Where Clauses are rather confusing. It will cause you a lot of effort and require understand the spec. There are no examples that I know of other than if you look at the source code. I believe in the source code related to events there is a hard coded filter and where clause that selects everything. This would give you a start point, but to make it actually work will probably require a lot of trial and error and there may be bugs here we don't even know about. You could dig around in here https://github.com/FreeOpcUa/python-opcua/blob/82f31c0d3d9f248a076ca821fa71a47c06f046a8/opcua/common/events.py#L116

SeongsuByeon, 2020-08-12:
@zerox1212  Thank you for your explanation. I have one question about unsubscribing.
As you know, I have to unsubscribe and subscribe nodes dynamically because nodes(traffic ships) are appearing and disappearing. My question is that is it possible to force it to stop subscribing?
We can keep monitoring items with time.sleep as below.  
   
`    
        def subscription(self):
        print("traffic subscription started")
        handler = SubHandler_trafficShip(self)
        sub = self.client.create_subscription(1000, handler)
        time.sleep(100000)
        sub.unsubscribe(handle)
        sub.delete()  `