FreeOpcUa/python-opcua #1114: Client-Server connection times out after a few minutes from Node-RED

sb934, 2020-08-11:
For simulation purposes, I have a server file written in python which refreshes new values every few seconds. I am trying to connect to this server running on my local-host, to read node values. The communication is SignedAndEncrypted (Basic256Sha256) policy ; where the server has it's own copy of certificate and private key attached and meanwhile I am using the default certificate, key pair of NodeRed OPCUA. I am able to connect and retrieve values for the first few minutes, however the connection fails after a few minutes and fails to reconnect. I am attaching the error log...
#### Error Log received on server.py

'''
Exception raised while parsing message from client, closing
Traceback (most recent call last):
File "/home/saksham/.local/lib/python3.7/site-packages/opcua/server/binary_server_asyncio.py", line 75, in _process_data
ret = self.processor.process(hdr, buf)
File "/home/saksham/.local/lib/python3.7/site-packages/opcua/server/uaprocessor.py", line 83, in process
msg = self._connection.receive_from_header_and_body(header, body)
File "/home/saksham/.local/lib/python3.7/site-packages/opcua/common/connection.py", line 309, in receive_from_header_and_body
chunk = MessageChunk.from_header_and_body(self.security_policy, header, body)
File "/home/saksham/.local/lib/python3.7/site-packages/opcua/common/connection.py", line 49, in from_header_and_body
decrypted = crypto.decrypt(data.read(len(data)))
File "/home/saksham/.local/lib/python3.7/site-packages/opcua/crypto/security_policies.py", line 172, in decrypt
return self.Decryptor.decrypt(data)
AttributeError: 'NoneType' object has no attribute 'decrypt'
'''
#### Error on node-red terminal

21:51:10.710Z :message_builder_base :109 Error client1 _decrypt_MSG : Sign and Encrypt : Invalid packet signature

AndreasHeine, 2020-08-11:
does it reconnect after restarting the flow?

sb934, 2020-08-11:
Yess it does. It repeats the same thing tho. Connects for a few minutes, returns the value and then disconnects throwing these error logs. 
Do you think there's an issue with the encryption, certificates?

AndreasHeine, 2020-08-11:
no if its with the certs it wouldnt work at all!

sb934, 2020-08-11:
That's exactly what I thought because the authentication is validated but then it disconnects. I am unable to pinpoint the issue here. Do you want me to share the server code?

AndreasHeine, 2020-08-11:
versions should be enought

sb934, 2020-08-11:
OPCUA version 0.98.12
NodeJS version 14.7
Node-RED version 1.1.2

AndreasHeine, 2020-08-11:
ok could take a while!

AndreasHeine, 2020-08-13:
could you have a look in node-red -> pallet -> ode-red-contrib-opcua -> version ?

AndreasHeine, 2020-08-14:
@sb934 pleas send me your client node as export!