FreeOpcUa/python-opcua #1115: python-opcua Client / Protocol Error occurs when trying to connect to Robot Controller

lhj5776, 2020-08-13:
Hello,

I would like to ask for help.
My python-opcua client cannot connect to the server(Robot Controller)

I already checked using UaExpert.
set up security settings
- made server accept the client certificate
- add the Server's certificate to UaExpert's trust list
- User authentication(ID / PASSWORD)
It worked fine.
I could connect to the server.
but when I tried to connect with python, it failed.

Below, I have written down my code and the message that appears when an error occurs.

_**python client code**_

from opcua import Client
import time

url = "opc.tcp://10.0.0.2:4880/Staubli"

client = Client(url)

#client.set_user("mainXXXXXX")
#client.set_password("XXXXXXXX")

client.connect()
print("Client Connected")

 
while True:
    #run_App = client.get_node("ns=5;s=Robot:Applications:Running Application")
    #running_App = run_App.get_value()
    
    #print(running_App)  

    time.sleep(1)

**_<message in cmd>_**

C:\Users\user\opcua>py client4.py
Received an error: MessageAbort(error:StatusCode(BadServiceUnsupported), reason:
None)
Received an error: MessageAbort(error:StatusCode(BadServiceUnsupported), reason:
None)
Protocol Error
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\ua_client.py", line 101, in _run
    self._receive()
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\ua_client.py", line 131, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callback
s in list are dict_keys([2])
Traceback (most recent call last):
  File "client4.py", line 16, in <module>
    **client.connect()**
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\client.py", line 277, in connect
    self.create_session()
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\client.py", line 389, in create_session
    response = self.uaclient.create_session(params)
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\ua_client.py", line 288, in create_session
    data = self._uasocket.send_request(request)
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\site-package
s\opcua\client\ua_client.py", line 83, in send_request
    data = future.result(self.timeout)
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\concurrent\f
utures\_base.py", line 430, in result
    raise CancelledError()
concurrent.futures._base.CancelledError

version : freeopcua-0.90.6   opcua-0.98.12

Thank you.

AndreasHeine, 2020-08-18:
your python client does not contain any security configuration! 

`client.set_security_string("Basic256Sha256,SignAndEncrypt,certificate-example.der,private-key-example.pem")`

or is the server configured to have no security policie ("None") and just user auth. ?

lhj5776, 2020-08-31:
> your python client does not contain any security configuration!
> 
> `client.set_security_string("Basic256Sha256,SignAndEncrypt,certificate-example.der,private-key-example.pem")`
> 
> or is the server configured to have no security policie ("None") and just user auth. ?

Dear AndreasHeine Thank you for your reply sir.

I would like to ask you a few questions to use what you have given me.

1) When I use 'client.set_security_string()' method, 
   Do I have to enter the path in here?

2) private-key.pem 
  Can I use it after bringing the private-key in the server(Robot controller)?

3) When the connection was the successful with UaExpert
 then, I used username/password in Authentication Settings.
 not Certificate & Private key.
 In this case, Can I use the method below?
#client.set_user("username-example")
#client.set_password("password-example")

[image]

4) client.activate_session()
 I'm trying to use this method, but It's hard to find any place related rources or examples.
 
Thank you for contribution. Have a Nice Day Sir.

AndreasHeine, 2020-08-31:
> I don't know exactly how the server configured because the server was developed by Robot Company : (

which endpoint did you choose in UaExpert if you connect (None / Basic256Sign256)?

---

> When I use 'client.set_security_string()' method,
> Do I have to enter the path in here?

correct!

---

> private-key.pem
> Can I use it after bringing the private-key in the server(Robot controller)?

not the private key... some server need the certificate in the trusted folder some not depend on the server

---

> When the connection was the successful with UaExpert
> then, I used username/password in Authentication Settings.
> not Certificate & Private key.
> In this case, Can I use the method below?
> #client.set_user("username-example")
> #client.set_password("password-example")

correct!

---

> client.activate_session()
> I'm trying to use this method, but It's hard to find any place related sources or examples.
> client.activate_session( username, password, 'certificate' )
> In this case, Does 'certificate' mean [path + certificate file name.der] located in server controller? or client PC ?
> Can I add that code after client.connect()? like this

if you just client.connect() the session should be established automatically

---

[HINT](https://github.com/AndreasHeine/SecurePythonOpcUaClient/blob/master/x509v3/client_x509v3.py)

lhj5776, 2020-08-31:
Thank you for answer !!

> > I don't know exactly how the server configured because the server was developed by Robot Company : (
> 
> which endpoint did you choose in UaExpert if you connect (None / Basic256Sign256)?

Security policy : Basic256Sign256
Message Security Mode : Sign
in UaExpert

Is my answer enough for what you asked?

>> When I use 'client.set_security_string()' method,
> Do I have to enter the path in here?
>
> correct!

I have additional questions.

client.set_security_string("Basic256Sha256, Sign, PATH/certificate-name.der, PATH/private-key-example.pem")

In this case, Is PATH located in server controller? or client PC ?
For example, 
client.set_security_string("Basic256Sha256, Sign, C:\user\opcua\certificate-example.der, C:\user\opcua\private-key-example.pem")  

There is Certificate in server's trusted folder.
So I will bring it in my client PC and use it.

Is this right?

AndreasHeine, 2020-08-31:
no typically its the other way around you need a own cert for your clientinstance wich has to be copied to the servers trusted folder, the way like you said is insecure because the server will have multiple client connection with the same cert... if the server checks cert content (ip, uri) and if it does not match your application its goeing to be rejected

lhj5776, 2020-08-31:
> no typically its the other way around you need a own cert for your clientinstance wich has to be copied to the servers trusted folder, the way like you said is insecure because the server will have [...]

Thank you for helpful comment.

I will try hard.

Have a nice Day !! :)

lhj5776, 2020-09-02:
> no typically its the other way around you need a own cert for your clientinstance wich has to be copied to the servers trusted folder, the way like you said is insecure because the server will have [...]

If I use only 'Sign'
client.set_security_string("Basic256Sha256, Sign, ??? , ??? ")

Do I need to enter a certificate and encryption key here?

AndreasHeine, 2020-09-02:
sure!