FreeOpcUa/python-opcua #1119: Error when getting value but not when getting other node data

dhiczc, 2020-09-01:
Hello 

I am trying to find out whether it is the way I call the OPC UA python code or is there something wrong with my OPC UA connection. Basically, I am able to get the node "sampletagname" and I am able to get other data such as display name but when I tried to get value or get data value, I get an error. Have anyone experienced this before?

```
client.connect()
client.load_type_definitions()  # scan server for custom structures and import them
sampleNode = client.get_node("ns=1;s=sampletagname")
print("Getting display name: ", sampleNode.get_display_name())
sampleNodeValue = sampleNode.get_value()
print("Getting data: ", sampleNodeValue)
```

Results:
Getting display name:  LocalizedText(Encoding:2, Locale:None, Text:sampletagname)

Traceback (most recent call last):
  File "c:/Users/XX/Desktop/OPCUA tests/OPCUA.py", line 67, in <module>
    sampleTagValue = sampleTag.get_value()
  File "C:\Users\XX\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\common\node.py", line 155, in get_value
    result = self.get_data_value()
  File "C:\Users\XX\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\common\node.py", line 164, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "C:\Users\XX\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\common\node.py", line 276, in get_attribute
    result[0].StatusCode.check()
  File "C:\Users\XX\AppData\Local\Programs\Python\Python38\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdUnknown: "The node id refers to a node that does not exist in the server address space."(BadNodeIdUnknown)

AndreasHeine, 2020-09-01:
are you sure its in namespace 1 ?

dhiczc, 2020-09-01:
yes it is in namespace 1.

I can see the values in UaExpert but only in the DA Plugin with the same tagname. I don't have the rights to browse through all the nodes in the server.

dhiczc, 2020-09-01:
It's Siemens OPC UA server (I think)

AndreasHeine, 2020-09-01:
could you post a screenshot from ua expert with the node attriebutes?

like:
[image]

dhiczc, 2020-09-01:
Unfortunately, I am not able to get to the specific node attribute in UaExpert, I have no access to that. What I have is only under documents

[image]

AndreasHeine, 2020-09-01:
if you wrote the nodeid right it should work! do you have the right permissions (user permissions)?

but i guess we havenot implemented views at this point it might be because of that but i am not sure...

dhiczc, 2020-09-01:
I am trying to find out whether I have the right permissions or if it is something with the library. I'm new to OPC UA and all I know is that I have read only access so far to a limited number of nodes. 

The NodeID is correct because I am able to get the display name of the node, the error only occurs when I try to call get_value. 

Could it be something with views, where maybe I have the permission for views but not read?

zerox1212, 2020-09-01:
If you are getting the value only via OPC DA, why should OPC UA work? It looks like there is some kind of OPC UA to DA middleware on your server. Seems strange.