FreeOpcUa/python-opcua #1121: BadTooManyOperations or BadTooManyVariables when subscribing

DominikSchwager, 2020-09-03:
Hello python opcua community, 

I am trying to subscribe to about 12500 variables and I am running into trouble. The code I am running works fine with 2000 variables, but with the full compliment it gives the BadTooManyOperations error message. When I split the operation up and run the script multiple times with slices of array, I can subscribe to 4000 before I run into BadTooManyVariables.
Using top in the shell I monitored load of the system, but it never got past 40% in either memory or cpu and most of that was taken up by the write operation to the mongodb.

Here is the relevant codesnippet:
`def main():
    config = Config()
    client = Client(config.OPCConnectionString)
    try:
        client.connect()
        print("Exploring data structure.")
        variables = explore_node(client.get_node("ns=2;s=Application"))
        print(len(variables))
        print("Start logging now.")

        handler = SubHandler(config)
        sub = client.create_subscription(1000, handler)
        print("Start subscribing")
        sub.subscribe_data_change(variables)
        print("Finish subscribing")`

As I explained "explore_node" just looks through the server, looking for variables and returning them. It finds just north of 12000.
I have tried several variants, like subscribing to the datachange in a loop (same behaviour) and splicing the variables and running the script multiple times (slightly different, as explained above).
I am attaching the error message for the single script approach below. According to my research this would be the preferred way anyway. I found examples online of at least one person subscribing successfully to 10000 variables and their code did not look different. 
[image]

Thanks for reading this wall of text and I appreciate any help I can get!

AndreasHeine, 2020-09-03:
does the server allow that massive amount of monitored items? there is a limit serverside i guess max_monitored_items !?

DominikSchwager, 2020-09-03:
> is it a Python OPC UA Server o_0 ?

No, it is not a python opc ua server. It is running on a Bosch Rexroth PLC. I am going to investigate about the limit.

swamper123, 2020-09-03:
Why 10k was chosen as Limit (in our project), I don't know. Perhaps like "it's big and small enough"... but this is maybe interessting for some future folks out there. ;)

But before we are looking for stuff in our Client, checking the server capabilities would be the best way.

AndreasHeine, 2020-09-03:
maybe you can workaround with a second client instance (mostly limits are per session!) but PLCs are most of the time really restrictive on that (lack of performance... to be honest its only there second job!)

DominikSchwager, 2020-09-04:
Thanks all, it was indeed a limitation of the PLC itself. The maximum number of items per subscription was set to 2000. We are going to change that.

closed 2020-09-04 by DominikSchwager