FreeOpcUa/python-opcua #1125: Most performant way to subscribe to multiple monitored items and export them to csv

JohannesM24, 2020-09-07:
With a single monitored item I am able to get data as fast as 390 Hz.

When adding a second one the data rate is halfed, and for both i just get about 190 HZ. Adding more items make it worse..

In my datachange_notifcation method I check for the type of monitored item and add it to my pandas dataframe. In scheduled time I also want to flush that data_frame to csv, thats why I am using locks before writing/flushing the data_frame.

Has anyone ever done this with this library?

AndreasHeine, 2020-09-07:
asyncio with asyncua (python-opcua in asyncio) could potentially be better in performance and you have to take care about buffering/queueing the notifications and write them bunch/stackwise in the file (which is SLOW) make sure you dont open and close the file to often! also use a ssd which does not contain the os to write the file onto!

JohannesM24, 2020-09-07:
Thank you for your quick comment, guess I should look into asyncua aswell then.

I think we want to monitor up to 10 items, but with the highest speed possible.

Right now i implemented it like this:

```   
 def datachange_notification(self, node, val, data):
        ts = data.monitored_item.Value.SourceTimestamp
        value = data.monitored_item.Value.Value.Value
        column = self.node_to_display_name_dict.get(node)
        lock.acquire()
        self.data_frame = self.data_frame.append({'timestamp' : ts, column : value}, ignore_index=True)
        lock.release()
```
and:

```
 def flush_df(self):
        print(' THE SIZE OF THE DATA FRAME IS ' + str(self.data_frame.shape[0]))
        lock.acquire()
        #print(self.data_frame)
        time_for_filename = self.data_frame["timestamp"].iloc[-1].to_pydatetime().isoformat(sep=' ', timespec='seconds').replace(":","-")
        
        self.data_frame.to_csv(file_prefix_name + time_for_filename + '.csv', mode='a')
        self.data_frame = self.data_frame.iloc[0:0]
        lock.release()
```
and in the main:
```
        t_start = time.time()
        t_end = t_start + runtime_seconds 
        while time.time() < t_end:
            time.sleep(save_cycle_seconds - ((time.time() - t_start) % save_cycle_seconds))
            handler.flush_df()
```

I think the main problem here is multiple notifications coming in to the Handler method and the append function being much too slow from pandas.

AndreasHeine, 2020-09-07:
go for asynio and use two queues! try to use as less time possible in the subhandler method (put it on a queue ore in a in memory sqlite3 database) try to not lock resources too long maybe use a generator to yield only one notificaton each cycle to avoid missing a notification!

closed 2020-09-07 by JohannesM24