FreeOpcUa/python-opcua #1126: Prevent auto printed messages upon connection

AndrewStuhr, 2020-09-09:
Hello

I'm seeing the following 3 lines printed after making an OPC connection:

[image]

Is there a way I can stop these from printing?

Thanks!

AndreasHeine, 2020-09-09:
these messages came from a logger! just for my understanding why do you want to achieve this?  sounds like you want to use "stdout" for something?

the thing is if you remove or overwrite the logger nothing gets logged anymore, the other way would be to make a virtual environment with a local version of python-opcua and comment these 3 lines out (if you just want to remove only these particular lines)

swamper123, 2020-09-09:
That realy looks like a logger message and I guess it is the one in ua_client.py in `create_session()` (no warrent).

Additionally to @AndreasHeine you may set the logger level to warning (but be aware that all debugging and info logs won't appear anymore, just warning- and error logs!), depending on your codecontext.

AndrewStuhr, 2020-09-09:
Thanks guys!  I'm using print() to pass data back to a webpage. The logging messages are nice, but giving me trouble with the return data format.  I'll try changing the warning level, and if that doesn't work, commenting out the lines.

zerox1212, 2020-09-09:
Not sure what you are doing, but using `print()` as output to a web server or something is probably not a good design.

AndreasHeine, 2020-09-09:
sounds like node-red pythonshell node 🤔 have you tought about using a websocket to pass data to the webpage?

AndrewStuhr, 2020-09-09:
Good to know guys.  I'm using IIS, calling python scripts with AJAX, and returning data with print().  I'm planning to switch to Flask at some point.  Thanks for the websockets idea, I'll look into that.  I'd love to hear any other ideas!