FreeOpcUa/python-opcua #1129: How to connect and get the value using node id

abhinavrawat27, 2020-09-12:
Hi 

I have OPC server which I have connected using ua expert. From ua expert software, I have got the nodeid of the parameters. Below is one of the node id:

    NS1|String|UI-A@supervision|exports-4-0-4:statistics-1:sigma2

I am getting the value of above node id using below code:

    sigma = client.get_node("NS=1;S=::UI-A@supervision|exports-4-0-4:statistics-1:sigma2")
    sigma_value = sigma.get_value()

But I am getting error:

    opcua.ua.uaerrors._base.UaStringParsingError: Could not find identifier in string NS=1;S=::UI-A@supervision|exports-4-0-4:statistics-1:sigma2

Can anyone please let me know how can I get the value of this node id. Thanks

AndreasHeine, 2020-09-12:
"ns=1;s=::UI-A@supervision|exports-4-0-4:statistics-1:sigma2"

abhinavrawat27, 2020-09-12:
Hi @AndreasHeine 

Thanks for your comment. I am now getting this error:

    opcua.ua.uaerrors._auto.BadNodeIdUnknown: "The node id refers to a node that does not exist in the server address space."(BadNodeIdUnknown)

Actually I have created a simulation server and have added the node id as 

    sigma = param.add_variable(NodeId('UI-A@supervision|exports-4-0-4:statistics-1:sigma2', 1), QualifiedName('Sigma', 3), 0)

This is reflecting as below photo in UA expert:
 
    https://we.tl/t-Fgn2lpZgOI

So when connecting it as 

    sigma = client.get_node("ns=1;s=::UI-A@supervision|exports-4-0-4:statistics-1:sigma2")
    sigma_value = sigma.get_value()

I am getting above error

AndreasHeine, 2020-09-12:
i dont like to click on such links... please drag and drop the Screenshot in the comment writing area it will be uploaded!

abhinavrawat27, 2020-09-12:
Apologies. Here is the screenshot.

[image]

AndreasHeine, 2020-09-12:
could you make a screenshot from ua experts attributes tab?

abhinavrawat27, 2020-09-12:
[image]

Here is the attributes tab

AndreasHeine, 2020-09-12:
cant realy reproduce... could be a typo in your node id!?

```python
from opcua import Server, ua

server = Server()
server.set_endpoint("opc.tcp://127.0.0.1:4840/")

server.get_objects_node().add_variable(ua.NodeId.from_string("ns=1;s=UI-A@supervision|exports-4-0-4statistics-1:sigma2"), "3:Sigma", 0)

server.start()
```

```python
from opcua import Client, ua

client = Client("opc.tcp://127.0.0.1:4840/")
client.connect()

print(client.get_node("ns=1;s=UI-A@supervision|exports-4-0-4statistics-1:sigma2").get_value())
```

closed 2020-09-12 by abhinavrawat27