FreeOpcUa/python-opcua #1131: Opcua Block Read and Block Write (BSEND/BRCV)

AdamOPCUA, 2020-09-16:
Dear community, 

while seeing an implemented opcua client, I realized that there is a function returning a variable block. So basically a "variable identifier block read" is needed as input while the opcua output is the requested block.

Moreover, there is a block write function, whereas a "variable identifier block write" is needed as an input. Now I'm trying to implement these block read/write functions in python, however i can not find relevant functions within the opcua library. All the functions included carry out operations on nodes.

Can you help me please?

swamper123, 2020-09-16:
What exactly is a variable block (I'm not familiar with that term)? Multiple varaible nodes? Multible variable values? A list of all attributes of a Node(s)?

AdamOPCUA, 2020-09-16:
Well, it is supposed to be a operation carried out on multiple nodes, manupulating the values. There should be the so called block-services which can transmit up to 64kByke each operation

swamper123, 2020-09-16:
So something like:
https://github.com/FreeOpcUa/python-opcua/blob/8ac613ae206f261eb01a9e4886280a2d2cc83e2f/opcua/client/client.py#L666
and
https://github.com/FreeOpcUa/python-opcua/blob/8ac613ae206f261eb01a9e4886280a2d2cc83e2f/opcua/client/client.py#L658
?

I can't find any other Methods which may fit to your request and I can't find "block-services" in the Specs either. 😕

AdamOPCUA, 2020-09-16:
Well enither did I :/ however, some of these functions might exist in c#  or java

swamper123, 2020-09-16:
Are the upper ones the stuff you are looking for?
If no and we don't find them neither here, nor in the specs you may have to build them on your own. : /

zerox1212, 2020-09-16:
What client has that? It sounds like something done for legacy support. I have never heard of "block read/write" in OPC UA specification.