FreeOpcUa/python-opcua #1132: OPC UA and Python

ksrangini05, 2020-09-16:
Hi all, I have learnt python for machine learning algorithms (basic level - numpy, scipy, matplotlib, regressions, knn, etc.). Now I got assigned to work in OPC UA with python coding in it. I am completely new to python codes for opc ua protocol to initiate communication between server and client. Please can anyone refer some links or tutorials to start with some samples  and also how to master the coding. Thanks in advance

AndreasHeine, 2020-09-16:
https://github.com/FreeOpcUa/python-opcua/tree/master/examples

AndreasHeine, 2020-09-17:
one thing to mention is that "opcua-binary" is a protocol, "opcua" is a communication standard which also describe how data is represented

swamper123, 2020-09-17:
For OPC UA itself I liked the introduction from [here](https://documentation.unified-automation.com/uasdkdotnet/2.1.0/html/L1OpcUaFundamentals.html).
The official Specs are online [here](https://reference.opcfoundation.org/v104/).
And dealing with OPC UA with our repo see examples, mentioned from Andreas (also they create many "Aaaaahhhhhh"-Effects).
If you like async stuff more than sync, you may want to switch to [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio).

AndreasHeine, 2020-09-22:
http://wiki.opcfoundation.org/index.php/UA_Capabilities

ksrangini05, 2020-10-16:
To connect the server, I am using Unified Automation UA Expert client tool.  While connecting the server program(with user authentication -username & password)  with UA client tool, I got error  as "Bad Certificate Entrusted". Then I checked with server with encryption example from github- python opcua examples( https://github.com/FreeOpcUa/python-opcua/tree/master/examples) which has certificate and private key in the same program. I am able to connect but still I am not getting any data.  Please can anybody help me on this?
1. What are the basic python functions which need to be in server program for client authentication? 
2. If any simple documents /tutorials available on this, please refer

Thank you

swamper123, 2020-10-16:
What does your server want? User/Password or a certificate?

ksrangini05, 2020-10-20:
> What does your server want? User/Password or a certificate?
I just want to include user authentication code in this server with encryption example from github- python opcua examples( https://github.com/FreeOpcUa/python-opcua/tree/master/examples). I dont understand the user manager examples in github( https://github.com/FreeOpcUa/python-opcua/tree/master/examples).