FreeOpcUa/python-opcua #1133: UA Server : duplication of node parent properties to childrens (deep)

TheAtlantis, 2020-09-18:
Hello, 

I have created an ua server instance (thanks for your work) and I have a "bug" in node.
i have created custom object type and load it with xml file (skel.xml joigned as zip).
[skel.zip](https://github.com/FreeOpcUa/python-opcua/files/5244660/skel.zip)

When object type "LOCATION" is instancied, all childrend of this instance repeat parent properties, I don't understand why.

Instance code :

```
network in networks : 
            # Instantiate Network object to UA Tree
            nodes = instantiate(self.crsono_folder, 
                       network_object_type, 
                       bname="{}".format(network.name), 
                       nodeid=ua.NodeId.from_string("ns=%d;s=%s;"%(self.namespace_index, "{}.{}".format(self.crsono_folder.nodeid.Identifier, network.code).upper())))

            if not nodes : raise ReferenceError("Unable to instantiate network ua object type")
            
            network_object = nodes[0]
            
            # Set Attributes values
            network_object.set_attribute(ua.AttributeIds.DisplayName, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(network.name)),varianttype=ua.VariantType.LocalizedText)))
            network_object.set_attribute(ua.AttributeIds.Description, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(network.name)),varianttype=ua.VariantType.LocalizedText)))
            
            # Set properties values
            network_name_var_id = network_object.get_child(["{}:id".format(self.namespace_index)])
            network_name_var_id.set_value(network.id)
            network_name_var_name = network_object.get_child(["{}:name".format(self.namespace_index)])
            network_name_var_name.set_value(network.name)
            network_description_var_node = network_object.get_child(["{}:description".format(self.namespace_index)])
            network_description_var_node.set_value(network.description)
            network_web_page_var_node = network_object.get_child(["{}:web_page".format(self.namespace_index)])
            network_web_page_var_node.set_value(network.web_page)
            
            # Get LocationGroup for Network
            locationGroups = LocationGroup.objects.filter(network = network)
            
            # Check if locationGroups exist for NETWORK
            if (locationGroups.exists() ) :
                # Create folder object
                locationGroups_folder = network_object.add_folder("ns=%d;s=%s;"%(self.namespace_index, "{}.LOCATION_GROUP".format(network_object.nodeid.Identifier).upper()), "LocationGroups")

            for locationGroup in locationGroups :
                # Instantiate locationGroup object to UA Tree
                nodes = instantiate(locationGroups_folder, 
                                       locationgroup_object_type, 
                                       bname="{}".format(locationGroup.name), 
                                       nodeid=ua.NodeId.from_string("ns=%d;s=%s;"%(self.namespace_index, "{}.{}".format(locationGroups_folder.nodeid.Identifier, locationGroup.code).upper())))
                
                if not nodes : raise ReferenceError("Unable to instantiate location ua object type")
                
                locationGroup_object = nodes[0]
                
                # Set Attributes values
                locationGroup_object.set_attribute(ua.AttributeIds.DisplayName, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(locationGroup.name)),varianttype=ua.VariantType.LocalizedText)))
                locationGroup_object.set_attribute(ua.AttributeIds.Description, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(locationGroup.name)),varianttype=ua.VariantType.LocalizedText)))

                # Set properties values
                locationGroup_var_id = locationGroup_object.get_child(["{}:id".format(self.namespace_index)])
                locationGroup_var_id.set_value(locationGroup.id)
                locationGroup_var_name = locationGroup_object.get_child(["{}:name".format(self.namespace_index)])
                locationGroup_var_name.set_value(locationGroup.name)
                locationGroup_var_order = locationGroup_object.get_child(["{}:order".format(self.namespace_index)])
                locationGroup_var_order.set_value(locationGroup.order)
                locationGroup_var_transportation_type = locationGroup_object.get_child(["{}:transportation_type".format(self.namespace_index)])
                locationGroup_var_transportation_type.set_value(locationGroup.transportation_type)
                locationGroup_var_type = locationGroup_object.get_child(["{}:type".format(self.namespace_index)])
                locationGroup_var_type.set_value(locationGroup.type)
[... 125 lines cut ...]
                                     
                                    # Set Attributes values
                                    equipement_object.set_attribute(ua.AttributeIds.DisplayName, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(equipement.name)),varianttype=ua.VariantType.LocalizedText)))
                                    equipement_object.set_attribute(ua.AttributeIds.Description, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(equipement.name)),varianttype=ua.VariantType.LocalizedText)))

                                    # Set properties values
                                    equipement_var_id = equipement_object.get_child(["{}:id".format(self.namespace_index)])
                                    equipement_var_id.set_value(equipement.id)
                                    equipement_var_code = equipement_object.get_child(["{}:code".format(self.namespace_index)])
                                    equipement_var_code.set_value(equipement.name)
                                    equipement_var_name = equipement_object.get_child(["{}:name".format(self.namespace_index)])
                                    equipement_var_name.set_value(equipement.name)
                                    equipement_var_type = equipement_object.get_child(["{}:type".format(self.namespace_index)])
                                    equipement_var_type.set_value(equipement.type)
                                    equipement_var_version = equipement_object.get_child(["{}:version".format(self.namespace_index)])
                                    equipement_var_version.set_value(equipement.version)
```

View with UAExpert client attached.
[image]
 

Can you help me ?

Thanks

AndreasHeine, 2020-09-18:
please make server.export_xml() and post content so i can see the references!

TheAtlantis, 2020-09-18:
> please make server.export_xml() and post content so i can see the references!

I, thank for your help, here export file from root node recursively.

[export.txt](https://github.com/FreeOpcUa/python-opcua/files/5245570/export.txt)

zerox1212, 2020-09-22:
Can you take a screenshot of UA Expert showing your base custom object type from which you are instantiating from?

TheAtlantis, 2020-09-22:
> Can you take a screenshot of UA Expert showing your base custom object type from which you are instantiating from?

Hi, here the custom object type view by UA Expert (LOCATION seem to be the beginning of error).

[image]
[image]

zerox1212, 2020-09-22:
Those screenshots are not what I was looking for. There is another part of the server where you can see the custom object types that are derived from BaseObjectType.

TheAtlantis, 2020-10-26:
Hello,
Problem solved, integrating types into types generates attribute inheritance.
The solution and therefore to create types at the root.

closed 2020-10-26 by TheAtlantis