FreeOpcUa/python-opcua #1134: Instanciated object display object type instead of its name

TheAtlantis, 2020-09-18:
Hello, 

when use Instanciate to instanciate object type the client (UA Expert) show object type name instead display name.
Sample code : 

```
                    location_object_type = locationgroup_object_type.get_child(["%d:LOCATION" % self.namespace_index])
                    # Instantiate location object to UA Tree
                    nodes = instantiate(locations_folder, 
                                           location_object_type, 
                                           bname="{}".format(location.name), 
                                           nodeid=ua.NodeId.from_string("ns=%d;s=%s;"%(self.namespace_index, "{}.{}".format(locations_folder.nodeid.Identifier, location.code).upper())))
                    
                    if not nodes : raise ReferenceError("Unable to instantiate location ua object type")

                    location_object = nodes[0]
                    
                    # Set Attributes values
                    location_object.set_attribute(ua.AttributeIds.DisplayName, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(location.name)),varianttype=ua.VariantType.LocalizedText)))
                    location_object.set_attribute(ua.AttributeIds.Description, ua.DataValue(ua.Variant(value=ua.LocalizedText("{}".format(location.name)),varianttype=ua.VariantType.LocalizedText)))
```

String in "location.name" is present in DisplayName in UAExpert but not in tree...

Display in client : 
[image]

Have you an idea for this problem ?

Thanks

AndreasHeine, 2020-09-18:
--> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-custom-object.py

AndreasHeine, 2020-09-18:
```python
    # Example 2 - create a new object type and a instance of the new object type
    #-------------------------------------------------------------------------------
    types = server.get_node(ua.ObjectIds.BaseObjectType)
    
    object_type_to_derive_from = server.get_root_node().get_child(["0:Types", 
                                                                   "0:ObjectTypes", 
                                                                   "0:BaseObjectType"])
    mycustomobj_type = types.add_object_type(idx, "MyCustomObjectType")
    var = mycustomobj_type.add_variable(0, "var_should_be_there_after_instantiate", 1.0) # demonstrates instantiate
    var.set_modelling_rule(True) #if false it would not be instantiated
    myobj = objects.add_object(idx, "MyCustomObjectA", mycustomobj_type.nodeid)
    #-------------------------------------------------------------------------------
```

TheAtlantis, 2020-09-18:
Hi, 

using  `myobj = objects.add_object(idx, "MyCustomObjectA", mycustomobj_type.nodeid)` solve the problem.

Thanks

closed 2020-09-18 by TheAtlantis