FreeOpcUa/python-opcua #1136: Maximum Number of Continuation Points / Invalid Type

aschaepper, 2020-09-29:
Hello there

In the course of a project here at my company, I used this opc ua sever implementation, in combination with our realtime os.
When I wanted to connect to the server it refuses to connect with a bad type mismatch error. However, by checking this the answer from our secondlevel support resp. dev team was that:

"The maximum number of ContinuationPoints is read out in the course of the UA_Connect call. The Pyton OPC UA server does not supply a valid data type for this node. However, the client library checks whether it is a UInt16, as per specification. Since this ist not the case, the error OpcUa_BadTypeMismatch is returned."

I tried other clients such as OPC UA Expert, and they were able to connect (most probably having the issue "in the background"). But our library in our RTOS is just way more strict and refuses to connect.

Could you add this in your implementation? Our dudes are kind of strict and don't want to make our library more forgiving. If you point me into the right direction in your code, I could maybe fix resp. add it myself. I'm a dev and I also know python, so if its not a super heavy fix I can do it and I will offcourse contribute  with a pul request.

Greetings

AndreasHeine, 2020-09-29:
how do you write the value? there are more then one way to do it: value, variant and datavalue siemens plcs for example accepts only variants and datavalues!

aschaepper, 2020-09-29:
Hello

I dont write any value. I just make a connection from our PLC and it's not even able to connect, because when there I's a connect, our clientlibrary checks for this "maximum number of ContinuationPoints", which should be provided by the server, and should be of type int16 as far as I understand.

This is what I found, please forgive me I'm not a OPC UA expert and just give you the infos I recieve from our dev team.
https://reference.opcfoundation.org/v104/Core/docs/Part4/7.6/

"Servers specify a maximum number of ContinuationPoints per Session in the ServerCapabilities Object defined in OPC 10000-5"

AndreasHeine, 2020-09-29:
no problem!

which version of python opcua did you use? latest?

according to specs it looks right! 
we generate the standard addresspace from spec. xml files of the opc foundation!

https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.2/

[image]

[image]

[image]

[image]

AndreasHeine, 2020-09-29:
https://raw.githubusercontent.com/OPCFoundation/UA-Nodeset/v1.04/Schema/Opc.Ua.NodeSet2.Part5.xml

```xml
  <UAVariable NodeId="i=3089" BrowseName="MaxBrowseContinuationPoints" ParentNodeId="i=2009" DataType="UInt16">
    <DisplayName>MaxBrowseContinuationPoints</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">i=2009</Reference>
    </References>
  </UAVariable>
```

zerox1212, 2020-09-29:
I think our server is defined correctly so I'm not sure why your client has a problem. In general being "strict" with OPC UA is going to cause you nothing but headaches. Not connecting due to something inconsequential like ContinuationPoints, which most servers probably won't even define anyways, doesn't sound very robust.

aschaepper, 2020-09-29:
"Not connecting due to something inconsequential like ContinuationPoints, which most servers probably won't even define anyways, doesn't sound very robust."

Couldnt agree more! Unfortunately I can't force them to change their minds :) or it has different reasons I don't know about why they wont change it.

AndreasHeine, 2020-09-29:
the question is why do they think it has to be an int16, the only way i can imagine is that it was in an pre 1.04 version of opcua!? so it might be compliante to an older opcua release!? but anything else would be not acceptable in my opinion. tbh they have to update to be compliant!!!

aschaepper, 2020-09-29:
I will rebuild a testcase, and go back to our dev department with your new informations given.

Ill keep you updaten, im sort of inbetween two parties here. Appreciate your great help!

ChUnterberger, 2020-10-05:
The issue here is that as you can see above, while the data type is UInt16 the value is unfortunately NULL. 

[image]

As such when the client requests the value attribute of the MaxBrowseContinuationPoints (i=2735), the Python OPC UA server responds with Variant Type: Null (0x00) as can be seen in a Wireshark trace.

[image]

This has been reported in the past for the opcua-asyncio library with the following issue but to the best of my knowledge has never been resolved in either library for the default server configuration. https://github.com/FreeOpcUa/opcua-asyncio/issues/85

According to OPC UA Part 5 - Information Model Release 1.04 Specification / 6.3.2 ServerCapabilitiesType the MaxBrowseContinuationPoints is mandatory and a UInt16.

_"MaxBrowseContinuationPoints is an integer specifying the maximum number of parallel continuation points of the Browse Service that the Server can support per session. The value specifies the maximum the Server can support under normal circumstances, so there is no guarantee the Server can always support the maximum. The client should not open more Browse calls with open continuation points than exposed in this Variable. The value 0 indicates that the Server does not restrict the number of parallel continuation points the client should use"_
https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.2/

To the best of my knowledge it is sufficient to set the value of MaxBrowseContinuationPoints in the server.
The chosen value of 10 is just a arbitrary value. I have not actually checked how many parallel browse requests the architecture would support.

`MaxBrowseContinuationPoints = server.get_node(ua.NodeId(2735))`
`MaxBrowseContinuationPoints.set_value(ua.Variant(10, ua.VariantType.UInt16))`

If you still have issues then please also set the ServiceLevel which is also mandatory and NULL on the PythonOPCUA server.
`ServiceLevel = server.get_node(ua.NodeId(2267))`
`ServiceLevel.set_value(ua.Variant(255, ua.VariantType.Byte))` 

_"ServiceLevel describes the ability of the Server to provide its data to the client. The value range is from 0 to 255, where 0 indicates the worst and 255 indicates the best. Part 4 defines required sub-ranges for different scenarios. The intent is to provide the clients an indication of availability among redundant Servers."_
https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.1/

AndreasHeine, 2020-10-05:
@ChUnterberger sure:
```python
MaxBrowseContinuationPoints = server.get_node(ua.NodeId(2735))
MaxBrowseContinuationPoints.set_value(ua.Variant(10, ua.VariantType.UInt16))
```
thats the right way to set it!

siemens checks server_status and service_level as connection monitoring so each vendor is a little different.
null is a valid value for "MaxBrowseContinuationPoints" / a UInt16 its just not defined -> has to be defined by the server-developer

but i dont know @aschaepper "clientlibrary" and why they expect a Int16 and not a UInt16 like in the Specs. defined and in our server implemented.

aschaepper, 2020-10-05:
Hi together

I will be able to look into this this week. It could also be that the error
is simply on our/my side. I will check and come back to you I promise!

We had the exact same issue, same error but it could be a coinsidence and
once we look into it its something else.

I just need to take another wireshark trace, send it to our guys and then I
see the respone. Given your provided informations I for sure have more
facts pointing to. As Mr Heine said it could be that we comply to an old
Version. In this case they might upgrade...

Pleade give me some time.

thank you all!

AndreasHeine, 2020-11-09:
@aschaepper somehow i missed your response... I apologize!

> I did not understand your first comment, sorry:

Python DataTypes are different then OPC UA DataTypes

if you just use set_value() and dont provide the VariantType we have to guess which type you mean (transition from dynamic typed language to a static typed one) this is not always the right one for example take the value 100 it could be multiple different opcua types: UInt16, UInt32, UInt64, Int16, Int32, Int64 even byte (many chances to get it wrong!)

https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/common/node.py#L195
https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/ua/uatypes.py#L766

thats the reason why its important to pass the VariantType!

### The Value of a OPC UA Variant can always be Null!
https://reference.opcfoundation.org/v104/Core/docs/Part6/5.1.6/

> Variants can be empty. An empty Variant is described as having a null value and should be treated like a null column in a SQL database. 

If you setup your Python OPCUA Server you need to initialize the ServerCapabilities with your values, we dont do that!

so unfortunatly your library is not compliant to the spec. in that point! You need to add some fallback defaults if a server provide ServerCapabilities with an empty Variant!

lg Andreas

aschaepper, 2020-11-17:
I did change my server, so its fine. On your side nothing is wrong all according standard! Thank you very much for your support!

closed 2020-11-17 by aschaepper