FreeOpcUa/python-opcua #1139: What could be the reason for only one client being able to make connection with server

abhinavrawat27, 2020-10-05:
Hi

I have written a opc ua client code to connect with opc ua server. This opc ua server is managed by some other manufacturer. Before deploying my code, I tested the connection with server using ua expert and everything seems to be working fine. I then deployed my opc ua client python code but wasn't able to connect to the server.

Later when I closed the connection from the ua expert, I was then able to make connection with server from the python code. I tried searching for why 2 clients cannot connect to the server at a time but I didn't get any results. Is there any particular reason why only one client is able to make connection to the server. ? Any suggestions. Thanks

AndreasHeine, 2020-10-05:
server settings max sessions to small.... look with ua expert under server -> server capabilitys there must be some information about!

abhinavrawat27, 2020-10-05:
@AndreasHeine Thanks I will check this and will update soon.