FreeOpcUa/python-opcua #1140: Is it possible a server callback before the ItemSubscriptionCreated?

miguelarturo, 2020-10-06:
I need to create a variable **after** a client asks for the subscription, I mean, the variable does not exist until a client asks for it, at this moment the variable is created.
This callback:
  server.subscribe_server_callback(CallbackType.ItemSubscriptionCreated, create_monitored_items)
returns item not found the first time as expected.

The question is if it is possible to intercept the subscription in order to create the variable an avoid this error.

swamper123, 2020-10-07:
You want to sub to a Node which doesn't exist and while you sub, this Node shall be created, if it doesn't exist (and I guess it only shall create predefined Nodes, not just every Node which may be subbed)?

What I recommened is, that you may create an UA Method which creates the Node and then you can sub to it as a Client. This way would be easier, instead of implementing Code into the subscribe or monitored items routine with checks if that Node is allowed to be created etc.

I am courius about the usecase, can you provide some backround informations?

Other ideas are welcomed.

swamper123, 2020-10-07:
Today I learned, that a Client is able to create Nodes on it's own. So you can make a function/method which is creating the Node on the server and if that was successfully, you can sub to that Node.
But this has to be implemented on client side itself.

AndreasHeine, 2020-10-07:
you should browse befor try to subscribe and if it does not exist i would implement a node-create-method (create node and fire a event) which the client calls after recv a node-created-event advantage of that pattern would be that the method provides a propper template (or format) and you can emit an event after creation and your clients just simply subscribe if the event arrives and are synced with the new node.

be carefull with dynamic addressspaces if your not 100% sure what your doing and what the consequences could be! you need to take care about node lifespan .... and more... otherwise you could run out of ressources ;)
another question is the linking of information if its static on serverside and its got written by the clients only its not a big deal but if the server also has to write its getting more complex

miguelarturo, 2020-10-07:
Let me explain briefly the use-case, it is to implement a communication driver with a PLC. 
A PLC has data areas to interchange information. The PLC data areas are addressed by the position (Modbus plus has 64K 16bits registers x 4 areas).
The OPC client connects with the OPC server and can read/write some registers, each register can be addressed by Area@Position. Each register has a numeric value corresponding to different field variables.

In a first attempt, I created an object for any PLC register in the OPC server, this creates a really large number of objects which produces communication time outs in the clients (UAExpert)

To solve this issue the right solution seems to be to create only the items to be monitored by the OPC clients. This solution also optimizes the communication performance between the OPC server and the PLC, because only the items to be monitored are interchanged.

The idea is to create the items which are well addressed and does not exist in the OPC server right when the client subscribes. 
The only way I found was using server.subscribe_server_callback(...), but as mentioned before this produces an error because the callback is executed after the server looks for the variable.

zerox1212, 2020-10-07:
I also think some UA methods to manage the PLC data mapping would be the best way. You could make method to add/update/delete a node that represents your PLC registers. After the client add the new node it can do a normal browse to check that it exists then subscribe as normal.

closed 2020-10-09 by miguelarturo

miguelarturo, 2020-10-09:
I can not create the items on the client-side, because the clients are out of my scope, belong to third-parties.
I solved creating all the possible variables in the OPC server at startup, this is not the ideal solution but is working.
Thanks all for your support.

ASolchen, 2021-07-21:
I'm am also looking for a way to do this. Client could be third-party so they would request a browse name dynamically. This is the way KEPServer OPC works. In the server config you define node paths that map to devices that use other protocols. Then when the client requests a the full node path, the server uses the string being requested to poll the end device. For example, the server code could have a client-side section connecting to a Rockwell PLC. The PLC uses ASCII tag names for its data. Say you have a tag called "FOO" so when an OPC client requests the node value "MyPLC.FOO" the server could poll the PLC for "FOO." If the tag is good, create an OPC node under "MyPLC" and update values to "MyPLC.FOO"