FreeOpcUa/python-opcua #1141: odeId of type GUID that is parsed from string can't be converted to binary python-opcua

paperinus, 2020-10-07:
The issue is the same as: https://github.com/FreeOpcUa/opcua-asyncio/issues/49
This issue was resolved with this push https://github.com/FreeOpcUa/opcua-asyncio/pull/50/files in the new repository asyncio but it's still in the python-opcua repository. The solution is the same could you fix it?
thank you

paperinus, 2020-10-07:
Just change line 365 (identifier = v) with: identifier = uuid.UUID(f"urn:uuid:{v}")
I did it on my workspace and everythings works fine

paperinus, 2020-10-12:
Hello?

swamper123, 2020-10-12:
Can you make a PR?

Bawwa-cmd, 2021-03-08:
Did anyone find a solution for this? :(

oroulet, 2021-03-08:
just made a test in opcua-asyncua and it works there. so you can use the sync api there

Bawwa-cmd, 2021-03-08:
After applying this to the code, can we read /write the values of GUID nodes as usual?(Using get_node and set_value)
@oroulet

oroulet, 2021-03-08:
please make a PR then

Bawwa-cmd, 2021-03-09:
I get the error called ; 'Node' object has no attribute 'get_node'. Could  you please help me out :( @oroulet