FreeOpcUa/python-opcua #1145: Unable to read data from node. Throwing opcua.ua.uaerrors._auto.Bad: The operation failed.(Bad) error.

sb934, 2020-10-12:
I am trying to read data from a Kepware server at regular intervals. I am able to read some tags in the address space; however some tags end up giving me a _opcua.ua.uaerrors._auto.Bad: The operation failed.(Bad)_ error. So I included a try/catch so that the code is not interrupted, skips the tags throwing Bad StatusCode. 
Check the following output (of 1 iteration)...

Node1: 333

ua.uaerrors._auto.Bad @  Node2
ua.uaerrors._auto.Bad @  Node3

Node4 : 0

Node5 : 1

Node6 : 9

Node7 : False

Something that doesn't add up is that if I open these Nodes (1,2,3) in UAExpert, the Python script is suddenly able to read values for those nodes too (like Node1 above). It takes a few seconds for UAExpert to get Good StatusCode for these nodes. (BadInitialWaiting -> Bad -> Good) . Similarly, if I close UAExpert the Python script starts catching Bad error on these nodes again... 

Is there a way I can repeatedly try reading a node value, before moving on to the next line of code? Or any other workaround this issue?

AndreasHeine, 2020-10-12:
try reading if status is not good raise error/except pass in a while loop and if reading is good break!?

sb934, 2020-10-12:
I have my Node Details in a JSON file. This is what I am doing... 

`if(len(config['Tags'])):
            for obj in config['Tags']:
                try:
                    node = client.get_node(obj['node_addr'])
                    node_value = node.get_value()
                    print(obj['tag'],':',str(node_value),'\n')
                    v1 = [obj['tag'],str(node_value),obj['msg_type'],0]
                    push_value(c, v1)
                except ua.uaerrors._auto.BadWaitingForInitialData:
                    print("ua.uaerrors._auto.BadWaitingForInitialData @ ", obj['tag'])
                    #pass 
                except ua.uaerrors._auto.Bad:
                    print("ua.uaerrors._auto.Bad @ ", obj['tag'])
                    #pass
else:
    print("No Tags to retrieve from server")`

AndreasHeine, 2020-10-12:
can you post a screenshot from ua expert of those nodes (addressspace and attributes)? 
have you tryed to subscribe the nodes, instead of polling, to get the data it much simpler because you will get a datachange notification if the node value change!?

sb934, 2020-10-12:
[image]

[image]

[image]

[image]

As mentioned earlier: As soon as I open these tags in UAExpert. The Python script is able to read values without any exceptions.

sb934, 2020-10-12:
@AndreasHeine you got anything mate?

AndreasHeine, 2020-10-13:
sorry took some sleep guess your in a different timezon 🤔

AndreasHeine, 2020-10-13:
try browse them before reading value!!?

sb934, 2020-10-13:
If I use the subscriber class instead of manually reading the value. It throws the same "Bad Status" error. I can try again though. Didn't get you on the browsing part..?

sb934, 2020-10-13:
What I don't understand is that how is UAExpert and Python script connected? Like why does the code stop giving me errors and starts reading the value as soon as I start reading the same nodes in UAExpert?

AndreasHeine, 2020-10-14:
just call get browse_name before you subscribe or read the node! there was something in the past but i can hardly remember... not sure if it was kepware.

LostInDarkMath, 2021-07-13:
I have exactly the same problem as @sb934. The suggested solution from @AndreasHeine does not work for me. 
Does someone have an idea how I can fix this?

zerox1212, 2021-07-13:
Sorry, best answer I can give is to avoid using Kepware. The server is notoriously bad.

LostInDarkMath, 2021-07-13:
Are there any alternatives to Kepware that you would recommend?

AndreasHeine, 2021-07-13:
the thing with the the statuscode of kepware tags is it is dependend on the connectionstatus from kepware to the datahost... had this in a prev project the only thing you can do is to react to the bad statuscodes and try to browse the tags up front most of the time it helps otherweise it depends on the driver in kepware when it starts to update the opc ua tags...

yetkinakyz, 2022-02-11:
Hi there, I have the same problem that I cannot read some nodes from Sinumerik 840D opcua server and get 'Operation failed.(Bad)' error. Also, I can read all nodes with Prosys opcua client. Did you find any solution?

KenSense, 2022-11-28:
I have the same problem with SINUMERIK OPC UA Server.
One specific node doesn't want to be read.
/Tool/Catalogue/toolIdent[u1,49] it's the last one of a list.

That's the same error as if I go one index too far in another working list.
UaExpert doesn't give an error I can read whatever index I want out of range and it will still say it's ok without return value.

schroeder-, 2022-11-28:
As you see UAExpert can only read it via Subscription, but not via ReadRequest. So this is something the SINUMERIK OPC UA Server doesn't support and not the error of this library.

KenSense, 2022-11-29:
I can indeed subscribe to it, Thanks.
_New data change event ns=2;s=/Tool/Catalogue/toolIdent[u1,49] None_