FreeOpcUa/python-opcua #1146: Question: FreeOpcUa Server - catch timeout or disconnect to a Client

EddyITE, 2020-10-13:
Hey there,
is there a way to catch an event where the connection to a client is lost? 
My Intent is to historize only unsent messages for a subscriber. 
When the connection is lost, the server starts historizing certain nodes and when reconnected stop historizing.
The client can then read the history Data to get the lost Data which couldn't be sent due to unstable network.

oroulet, 2020-10-13:
polling seems to be the only reliable method

EddyITE, 2020-10-13:
How would I realize that? 
Not sure with what method I can communicate with the Client via the Server or which server attributes to check the connection state.

oroulet, 2020-10-13:
sorry I misread the question. There is no out of the box solution. Ideallly you would expand the CallbackService stuff which exists in opcua-asyncio to call some callbacks at connect and disconnect in binary_server.py