FreeOpcUa/python-opcua #1150: Change in StatusCode in server side is not notified to the clients

miguelarturo, 2020-10-20:
It seems when only the StatusCode is changed, the clients do not receive any update.

#this change will be notified to the clients
**value=value+1**
data = ua.uatypes.DataValue(variant=value, status=ua.uatypes.StatusCode(ua.StatusCodes.Good))
OPCVar.set_value(data) 

time.sleep(2)

 #when only change the status code, the client does not recieve the event
data = ua.uatypes.DataValue(variant=value, status=ua.uatypes.StatusCode(**ua.StatusCodes.Bad**))
OPCVar.set_value(data)

If the client, re-reads the item, the status code will reflect the correct status.
Anyway, I'm not sure if this is the right way to modify the status code.

AndreasHeine, 2020-10-20:
according to the specs clients shouldnt use values with bad statuscodes https://reference.opcfoundation.org/v104/Core/docs/Part4/7.34.1/

AndreasHeine, 2020-10-20:
also according to the specs (https://reference.opcfoundation.org/v104/Core/docs/Part4/5.12.1/) "The Value Attribute is used when monitoring Variables. Variable values are monitored for a change in value or a change in their status."

maybe you have subscribed the datachange (valuechange) and not the statuschange?

```python
class SubHandler(object):
    """
    Subscription Handler. To receive events from server for a subscription
    This class is just a sample class. Whatever class having these methods can be used
    """

    def data_change(self, handle, node, val, attr):
        """
        Deprecated, use datachange_notification
        """
        pass

    def datachange_notification(self, node, val, data):
        """
        called for every datachange notification from server
        """
        pass

    def event_notification(self, event):
        """
        called for every event notification from server
        """
        pass

    def status_change_notification(self, status):
        """
        called for every status change notification from server
        """
        pass
```

AndreasHeine, 2020-10-21:
```python
    def subscribe_data_change(self, nodes, attr=ua.AttributeIds.Value, queuesize=0):
        """
        Subscribe for data change events for a node or list of nodes.
        default attribute is Value.
        Return a handle which can be used to unsubscribe
        If more control is necessary use create_monitored_items method
        """
        return self._subscribe(nodes, attr, queuesize=queuesize)
```

AndreasHeine, 2020-10-21:
@oroulet if we sub to the attr value according to the specs we need to return the DataValue (value, status and timestamps)
https://reference.opcfoundation.org/v104/Core/docs/Part4/7.20.2/

```python
    def _call_datachange(self, datachange):
        for item in datachange.MonitoredItems:
            with self._lock:
                if item.ClientHandle not in self._monitoreditems_map:
                    self.logger.warning("Received a notification for unknown handle: %s", item.ClientHandle)
                    self.has_unknown_handlers = True
                    continue
                data = self._monitoreditems_map[item.ClientHandle]
            if hasattr(self._handler, "datachange_notification"):
                event_data = DataChangeNotif(data, item)
                try:
                    self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
                except Exception:
                    self.logger.exception("Exception calling data change handler")
            elif hasattr(self._handler, "data_change"):  # deprecated API
                self.logger.warning("data_change method is deprecated, use datachange_notification")
                try:
                    self._handler.data_change(data.server_handle, data.node, item.Value.Value.Value, data.attribute)
                except Exception:
                    self.logger.exception("Exception calling deprecated data change handler")
            else:
                self.logger.error("DataChange subscription created but handler has no datachange_notification method")
```

possible fix:
```python
    def _call_datachange(self, datachange):
        for item in datachange.MonitoredItems:
            with self._lock:
                if item.ClientHandle not in self._monitoreditems_map:
                    self.logger.warning("Received a notification for unknown handle: %s", item.ClientHandle)
                    self.has_unknown_handlers = True
                    continue
                data = self._monitoreditems_map[item.ClientHandle]
            if hasattr(self._handler, "datachange_notification"):
                event_data = DataChangeNotif(data, item)
                try:
                    self._handler.datachange_notification(data.node, item.Value, event_data) #<-----------------------------
                except Exception:
                    self.logger.exception("Exception calling data change handler")
            elif hasattr(self._handler, "data_change"):  # deprecated API
                self.logger.warning("data_change method is deprecated, use datachange_notification")
                try:
                    self._handler.data_change(data.server_handle, data.node, item.Value, data.attribute) #<-----------------------------
                except Exception:
                    self.logger.exception("Exception calling deprecated data change handler")
            else:
                self.logger.error("DataChange subscription created but handler has no datachange_notification method")
```

@miguelarturo unfortunately at this moment we dont trigger a datachangeevent on status change!

oroulet, 2020-10-21:
@AndreasHeine we also send the entire even_data which should contain everything

miguelarturo, 2020-10-21:
I found as you mentioned the client I programmed in fact can receive the event when the Status is modified, the status code is right and the value received is null.
This is enough for my requirements.

Anyway, I'd like to mention I was debugging using the application UaExpert,  it seems only to subscribe to the value change.

Thank you very much Andreas for your answers.

closed 2020-10-21 by miguelarturo