FreeOpcUa/python-opcua #1151: Parsing of NodeID of type String fails if the string identifier contains ";"

Musiame, 2020-10-21:
Eg. : This NodeId is not parsed correctly: `"ns=1;s=V40;0;3"`

**Solution**
add `, 1` in _uatypes.py_ to split the string only at the first occurrence of `;` :
```
@staticmethod
def _from_string(string):
    l = string.split(";", 1)              # add   ', 1'
    identifier = None
    namespace = 0
    ntype = None
```

oroulet, 2020-10-21:
but is your example over valid? it looks really like something that is made to break things and should not be allowed by spec (but it might if they screwed up things)

Musiame, 2020-10-21:
This is a real life example. Don't know if it conforms to the specs, but I had to make the change is uatypes.py to make it work.

oroulet, 2020-10-21:
OK. but we need to make 100% sure this matches the spec and the second ";" is never used in some corner situation

AndreasHeine, 2020-10-21:
the specs are not clear in that point they only say:

https://reference.opcfoundation.org/v104/Core/docs/Part6/5.2.2/

> 5.2.2.4 String ToC
> All String values are encoded as a sequence of UTF-8 characters without a null terminator and preceded by the length in bytes.

which would include ";" but i have never seen a server with a IdentifierType String and ";" in the Identifier.
from which vendor is the opcua server or which stack?

Musiame, 2020-10-21:
In the spec OPC 10000-3, I've found:

> 8.2.4 Identifier type
> Identifier values of IdType STRING_1 are [String value] restricted to 4 096 characters.

and

> 8.32 String
> This Built-in DataType defines a Unicode character string that should exclude control characters
> that are not whitespaces.

But I agree it's not very clear. Can we really use any utf-8 char in the string identifier ?

[Dewesoft](https://dewesoft.com/) opc ua servers uses string identifiers containing ";".

oroulet, 2020-10-21:
What we should look at is th description of the string format of a nodeid. Not sure where it is...

AndreasHeine, 2020-10-21:
we should allow ";" in the Identifier!
[image]

i test it with a siemens variable after lunch ;)

Musiame, 2020-10-21:
I was not able to find a clear specification for the string identifier.

Nonetheless, the correction I propose does not change anything for the normal case without ";" in identifier.
And the purpose of the `split` function here is to split `ns=` and `s=` so I don't see a problem in modifying this code.

oroulet, 2020-10-21:
great make a pull request

AndreasHeine, 2020-10-21:
Siemens also allows ";"!
[image]

> And the purpose of the split function here is to split ns= and s= so I don't see a problem in modifying this code.

make sure the other IdentifierTypes arent affected!

zerox1212, 2020-10-21:
Should be easy to add a test for this in case there is a problem with using control characters in the future.

swamper123, 2020-10-26:
So the string should just be checked if:
- it contains control characters
- is smaller than 4096 bytes
- is UTF-8
The rest seems to be irrelevant.