FreeOpcUa/python-opcua #1153: User Authentication in Python OPCUA

ksrangini05, 2020-10-21:


zerox1212, 2020-10-21:
I guess we don't have an example for it, but if I remember correctly, you need to implement a `UserManager`.
https://github.com/FreeOpcUa/python-opcua/blob/913325635c9fa38b26c8b6227e9a0a6d3582f1bb/opcua/server/user_manager.py

Maybe you can make a server example for this and submit a PR.

AndreasHeine, 2020-10-21:
```python
try:
    from opcua import ua, uamethod, Server
    from opcua.server.user_manager import UserManager
    from time import sleep
except ImportError as e:
    print(e)

users_db =  {
                'user1': 'pw1'
            }

def user_manager(isession, username, password):
    isession.user = UserManager.User
    return username in users_db and password == users_db[username]

if __name__ == "__main__":
    """
    OPC-UA-Server Setup
    """
    server = Server()

    endpoint = "opc.tcp://127.0.0.1:4840"
    server.set_endpoint(endpoint)

    servername = "Python-OPC-UA"
    server.set_server_name(servername)
    address_space = server.register_namespace("http://example.net/UA")
    
    uri = "urn:opcua:python:server"
    server.set_application_uri(uri)
    
    server.load_certificate("cert.pem")
    server.load_private_key("key.pem")
    server.set_security_policy([
                                    # ua.SecurityPolicyType.NoSecurity,
                                    # ua.SecurityPolicyType.Basic128Rsa15_Sign,
                                    # ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
                                    # ua.SecurityPolicyType.Basic256Sha256_Sign,
                                    ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt
                                ])
    policyIDs = ["Username"]
    server.set_security_IDs(policyIDs)
    server.user_manager.set_user_manager(user_manager)

    """
    OPC-UA-Modeling
    """
    root_node = server.get_root_node()
    object_node = server.get_objects_node()
    server_node = server.get_server_node()

    """
    OPC-UA-Server Start
    """
    server.start()

    try:
        while 1:
            sleep(1)
    except KeyboardInterrupt:
        server.stop()
```

ksrangini05, 2020-10-21:
I have changed  the security policy to  "No security" because UA expert tool allows either certificate or user authentication. Its working.  I really appreciate your help. Thank you.

AndreasHeine, 2020-10-21:
@ksrangini05 please dont mix up 

AUTHENTIFICATION which allows to auth with username and password or also with a authentification certificate:
```python
    policyIDs = ["Username"]
    server.set_security_IDs(policyIDs)
    server.user_manager.set_user_manager(user_manager)
```

MESSAGESECURITY means encryption via key and cert:
```python
    server.load_certificate("cert.pem") #this has nothing to do with authentification 
    server.load_private_key("key.pem") #this has nothing to do with authentification 
    server.set_security_policy([
                                    # ua.SecurityPolicyType.NoSecurity,
                                    # ua.SecurityPolicyType.Basic128Rsa15_Sign,
                                    # ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
                                    # ua.SecurityPolicyType.Basic256Sha256_Sign,
                                    ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt
                                ])
```

ksrangini05, 2020-10-26:
> @ksrangini05 please dont mix up
>
> AUTHENTIFICATION which allows to auth with username and password or also with a authentification certificate:
>
> ```python
>     policyIDs = ["Username"] [...]

OK..thank you

ksrangini05, 2020-10-27:
- If I want to implement message security with key and certificate, can I use open ssl for it ?

AndreasHeine, 2020-10-27:
sure!

mathewssoris, 2020-10-31:
Is it possible to manage the certificates and private key in windows certificate manager and authenticate directly from there?

AndreasHeine, 2020-10-31:
https://reference.opcfoundation.org/v104/Core/docs/Part2/8/

mathewssoris, 2020-10-31:
Thank you for the link. Do you have any examples using FreeOpcUa library which uses certificates directly from windows certificate manager?
Because most of the examples i found are based on certificates stored in some local folders.

AndreasHeine, 2020-10-31:
No sorry!

ksrangini05, 2020-11-04:
**_Data Logging_**
I am using history.py to store the real time data and using UA expert tool to view them in two options - Data logger view and History Trend view. I want to store the data continuosly in excel/csv format. Can anyone help on this? (at any point of time,  i need to show the historical data for last 7days)

AndreasHeine, 2020-11-04:
> Data Logging
> I am using history.py to store the real time data and using UA expert tool to view them in two options - Data logger view and History Trend view. I want to store the data continuosly in [...]

@ksrangini05 a little off topic! sound like a general python topic...

JSkrat, 2021-04-05:
for opcua-0.98.13 I tried example from the comments, and opcua-client just logs in into the server without any username-password. Is the example still actual?