FreeOpcUa/python-opcua #1154: How to use custom structure as InputArgument in method?

AndreyKl4, 2020-10-27:
Hi all!

i am a very new with OPC UA. Perhaps, my question is a stupid one but for me, it is really hard nut to crack=) I have defined my own structure as in example(https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-create-custom-structures.py) and after that i tried to set my custom structure as an InputArgument in my method:

       basic_var = server.nodes.objects.add_variable(ua.NodeId(namespaceidx=nodeid), 'BasicStruct',
                                                                ua.Variant(None, ua.VariantType.Null),
                                                                datatype=basic_struct.data_type)
        basic_var = ua.Argument()
        basic_var.Name = 'Test_structure'
        basic_var.DataType = basic_struct.data_type
        basic_var.ValueRank = -3 
        basic_var.ArrayDimensions = 0

     
        node_methods.add_method(nodeid, 'simple_callback', call_back ,[basic_var], [])  

In UaExpert Client I can monitor my structure as InputArgument of my method but when  i click ''...'' i see empty ''value'' and '' name'' fields. Perhaps smb have already faced such problem and can help me with this issue=)

Thanks in advance!!!

Andrey

AndreasHeine, 2020-10-27:
typically you pass the int value of the varianttype into the list like:

```python
[ ua.VariantType.UInt16, ua.VariantType.UInt32, ua.VariantType.Float ]
```

i am still looking in the opcua specs about that topic...

zerox1212, 2020-10-27:
I do not think using extension objects with methods is well tested and I'm not sure how the associated python function will react.

AndreyKl4, 2020-10-28:
@AndreasHeine  @zerox1212 Thanks for answers. As far as i understood it is a bit difficult to use such structure as input argument in method:
        basic_struct_name = 'basic_structure'
        basic_struct = ua_server.create_structure(basic_struct_name)
        basic_struct.add_field('var1', ua.VariantType.Int32)
        basic_struct.add_field('var2', ua.VariantType.Boolean)
        basic_struct.add_field('var3', ua.VariantType.String)

        # add an advance structure which uses our basic structure
        nested_struct_name = 'nested_structure'
        nested_struct = ua_server.create_structure(nested_struct_name)
        nested_struct.add_field('var4', ua.VariantType.String)
        nested_struct.add_field('array', ua.VariantType.Int32)
        # add simple structure as field
        nested_struct.add_field('Field', basic_struct)

        basic_var = server.nodes.objects.add_variable(ua.NodeId(namespaceidx=nodeid), 'BasicStruct',
                                                            ua.Variant(None, ua.VariantType.Null),
                                                            datatype=basic_struct.data_type)
        basic_var = ua.Argument()
        basic_var.Name = 'Test_structure'
        basic_var.DataType = basic_struct.data_type
        basic_var.ValueRank = -3 
        basic_var.ArrayDimensions = 0

       node_methods.add_method(nodeid, 'simple_callback', call_back ,[basic_var], []) 

For my project I need to generate such nested structures in my server and after that to use them as inputs arguments in my methods. If i use all input arguments with out nested structure (flat), it will be difficult to find out in Client which parameter is responsible for what, because of big number of input arguments=( That is why i want to create first of all a nested structure and after that to use it as input argument. Perhaps smb has an idea in what way i can solve it=)

AndreasHeine, 2020-11-06:
@AndreyKl4
@zerox1212 

https://github.com/FreeOpcUa/python-opcua/pull/1158

lmundeja, 2020-11-18:
We have tried this way and it worked successfully. You can try this. 
`import os
import random
import sys
import time
import uuid
from datetime import datetime
from threading import Thread

from opcua import Client, Server, ua, uamethod
from opcua.common.type_dictionary_buider import (DataTypeDictionaryBuilder,
                                                 get_ua_class)

from OurProductDataType_Lib import OurProduct

@uamethod
def doSomethingwithModule(parent, new_product) : # OPC Ua method defined in your server to be consumed by client.
    print("dosomethingwithModule method in my server called with parameters: ")
    print("*********************************************")
    print(new_product)
    ###### Now we got the data from one of our clients who called this method with the parameter new_product. 
    
    ## By Calling this function DoSoemthingwithModbus with sending data received as parmater we can ...
    #.. call a function to do something in machine using Modbus communciation. 
    DoSomethingwithModbus(new_product)
  
    return "It works..."

class OPCUA_Client():
    
    #Create a new opcua client
    def __init__(self, endpoint,moduleName):
        self.client = Client(endpoint)
        self.moduleName= moduleName
    
    def __enter__(self):#Connect to Server

        self.client.connect()
        print("Client Connected....")
        self.mynamespace_idx = self.client.get_namespace_index("http://hs-emden-leer.de/OurProduct/") # Do not change this!!!
        self.root = self.client.get_root_node()
        self.obj = self.root.get_child(["0:Objects", "{0}:{1}".format(self.mynamespace_idx,self.moduleName)])
        self.client.load_type_definitions()  # scan server for custom structures and import them
        return self
       
    def SendData(self,data,methodName): #Method to send the received object from another server to next server where you are now connected as client
        res = self.obj.call_method("{0}:{1}".format(self.mynamespace_idx,methodName), data)
        print("Method answer is: ", res)
        return "ok done"    
    
    def __exit__(self, exc_type, exc_val, exc_tb):#Exit the server
        print("Disconnecting....")
        self.client.disconnect()

class OPCUA_Server(OurProduct):

    def __init__(self, endpoint, name):
        #Configuration
        print("Init", name, "...")
        self.name = name
        self.server = Server ()
        self.my_namespace_name = 'http://hs-emden-leer.de/OurProduct/' # Do not change this!!!
        self.my_namespace_idx = self.server.register_namespace(self.my_namespace_name)
        self.server.set_endpoint(endpoint)
        self.server.set_server_name(name)
       
        #Add new object - MyModule
        self.objects = self.server.get_objects_node()
        self.mymodule = self.objects.add_object(self.my_namespace_idx, "MyModule") #Mandotry please change the module name here to reflect your servers module.

[... 60 lines cut ...]

def DoSomethingwithModbus(data): ## Keep this function generic and can call when there ia an activity to be done by Machine using Modbus.

    try:
        print("Data Recived in Modbus Fucntion  ->>" )
        print(data)
        
        ### Write your ******!!!!!!!! Modbus !!!!!!!******** code to interact with machine here
      
        ### After that manipulate this data object(recived in paramter) and send it to next fucntion SendDataUsingOPCAUAClient as parameter. 
        for i in range(len(data.PathStack)):
            if data.PathStack[i].NameOfStation == "Input": ## This is the code for setting the step in PathStack wrt to your module to True.
                 data.PathStack[i].IsDoneSuccessful=True

        SendDataUsingOPCAUAClient(data) ## Calling the fucntion to connect client and send data
    except:
            print("Unexpected error:", sys.exc_info()[0])
            raise
   

if __name__ == '__main__': #Main run of your program

    
     StartOPCUAServer() #Start your OPC server by calling this method
`

lmundeja, 2020-11-18:
I have implemented custom structures in one of my projects. 
You can have a look and check if that helps you. 
[https://github.com/lmundeja/OPCUA-Webshop-MiniFactory/tree/main/OPCUA_Modbus](url)