FreeOpcUa/python-opcua #1157: Read value from CustomObject

ArkasDev, 2020-11-03:
I would like to read the values I receive from a client by call. I have linked the method and expect the values as parameters of the method. How can I get the value from the custom object as for the basic types?

```
def get_result_by_id(self, parent, result_id, timeout):
  print(timeout)
  print(timeout.Value)
  print(result_id)
  print(result_id.Value)
```
>Variant(val:12,type:VariantType.Int32)
12

>Variant(val:ResultIdDataType(Id: 1),type:VariantType.ExtensionObject)
ResultIdDataType(Id: 1)

AndreasHeine, 2020-11-03:
do you want to read a custom object type or a custom structure (extension object)?

AndreasHeine, 2020-11-03:
i am not sure if this is what you are asking for but if you load_type_definitions() the custom types are converted to python classes

the big question for me is how is the method defined? XML? i would expect that the inputparameter is a extension object!? maybe you can share some more details!: 
-which method? 
-input args / output args?

[image]

ArkasDev, 2020-11-03:
Thanks for your quick response. OK, there already exists a XML file, which contains the specifications. Now I want to implement functions specified in the XML on the server. On the one hand the function expects a complex data type and the function returns a complex data type (this is called extension object in the OPC UA standard am I right?). So my question is, how can I use the complex data type in the code like a class/type?

I would like to implement something like below. This function is called when the client sends the call.
```
(Pseudocode)
get_result_by_id(result_id, timeout):
  id = result_id.id
  ...
  result_data_type.id = 2
  result_data_type.isPartial = true
  ...
  return [handle, result_data_type, 12]
```
 

[image]

AndreasHeine, 2020-11-03:
> On the one hand the function expects a complex data type and the function returns a complex data type (this is called extension object in the OPC UA standard am I right?).

YES!

have not tested it but i would:

```python
def method(input_extobj, timeout):
    return [
        ua.Variant(out_extobj, ua.VariantType.ExtensionObject),
        ua.Variant(Error, ua.VariantType.Int32),
    ]
```
importent thing is you need to return a list of variants!

AndreasHeine, 2020-11-03:
hey i played a little around:
-for extension objects you dont need to use variant but for the nonExtensionObjects

```python
from opcua import Server, ua, uamethod

@uamethod
def callback(parent, in_extobj):
    print(in_extobj)
    out_extobj = ua.uaprotocol_auto.AxisInformation()
    out_extobj.EURange.High = 200.0
    print(out_extobj)
    return [
        out_extobj
    ]

server = Server()

idx = server.register_namespace("TEST")

# server.import_xml("Opc.Ua.Di.NodeSet2.xml")
server.load_type_definitions()

obj = server.get_objects_node()

ext_obj = server.get_node("ns=0;i=12079")

inarg_extobj = ua.Argument()
inarg_extobj.Name = "myExtObj_In"
inarg_extobj.DataType = ua.NodeId(12079, 0)
inarg_extobj.ValueRank = -1
inarg_extobj.ArrayDimensions = []

outarg_extobj = ua.Argument()
outarg_extobj.Name = "myExtObj_Out"
outarg_extobj.DataType = ua.NodeId(12079, 0)
outarg_extobj.ValueRank = -1
outarg_extobj.ArrayDimensions = []

method_parent = obj.add_object(idx, "Parent")
method_node = method_parent.add_method(
    idx, 
    "Method", 
    callback, 
    [
        inarg_extobj
    ], 
    [
        outarg_extobj
    ]
)

if  __name__ == "__main__":
    server.start()
    while 1:
        pass
```

here you can find all importet structures/generated classes
[image]

ArkasDev, 2020-11-05:
Thanks for your contribution @AndreasHeine. My problem is still that the parameter is wrapped as below. I want simply access the properties of `result_obj`, but I do not understand how that works. 

```
def get_result_by_id(self, parent, result):
  print(result) 
  # Output: Variant(val:ResultIdDataType(Id: 1),type:VariantType.ExtensionObject)

  result_obj = ua.Variant(result, ua.VariantType.ExtensionObject)
  print(result_obj.Value) 
  # Output: ResultIdDataType(Id: 1)

  # Something like: print(result_obj.Id)
  # Output: 1
```

ArkasDev, 2020-11-11:
Thanks, it helped me a lot

closed 2020-11-11 by ArkasDev