FreeOpcUa/python-opcua #1159: BadWriteNotSupported

Kronosson, 2020-11-05:
I'm completely new to OPC UA and i'm having a problem with my Client code.
I'm trying to write a new value on an existing node in my Siemens S7 1200. Reading it is easily done, but changing the value always gives me the following error:

> BadWriteNotSupported: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)

My code looks like this

```
from opcua import Client
from opcua import ua

client = Client("opc.tcp://localhost:4840")
client.connect()
 
var = client.get_node("ns=4;i=27")  
value = var.get_value()                         
print("Initial value {}".format(value))

var.set_value(500)                              
print ("New value {}".format(var.get_value()))
```

After this issue I searched a bit and found a few others who had similar problems, so tried adding in the following lines, as it seemed to fix the problem for some people. 

```
dv = ua.DataValue(ua.Variant(500, ua.VariantType.Int16))
dv.ServerTimestamp = datetime.datetime.now()
var.set_value(dv)
```

Unfortunately i'm still getting the same error message.

AndreasHeine, 2020-11-05:
it should work with datavalue! 
is the plc clock set to correct date and time?

```python
from opcua import Client, ua

url = "opc.tcp://192.168.10.1:4840"
client = Client(url)
client.session_timeout = 30000
client.connect()
print(f"Connected: {url}")

node = client.get_node('ns=3;s="TEST"."TestUint"')

print("before: ", node.get_value())
node.set_value(ua.DataValue(ua.Variant(5, ua.VariantType.UInt16)))
print("after: ", node.get_value())

client.disconnect()
print("Disconnected")
```
```
Connected: opc.tcp://192.168.10.1:4840
before:  200
after:  5
Disconnected
```

AndreasHeine, 2020-11-05:
`dv.ServerTimestamp = datetime.datetime.now()`

this is done from the set_value() method!

https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/common/node.py#L195

https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/common/node.py#L216

Kronosson, 2020-11-05:
Thank you very much. That solved this problem. 
Now i have the next. The Value is not set to the new value. There is no difference in the operation of my device.
I checked, if the nodes are set to writeable in TIA Portal and they are. But i can't change the values, neither with my script, nor with UA Expert. 
UA Expert even says, the write to node was succesful.
[image]

Is it possible, that something overrides my new value?

AndreasHeine, 2020-11-05:
PLC itself maybe! Does the Value get written Cyclic somewhere in the PLC-Program? Check Cross-References in TIA.

Kronosson, 2020-11-10:
You were right about that. Thank you very much.
The PLC overwrites every change.  A collegue of mine was able to fix the issue within in PLC. Now i'm able to change the values, as it should be.

jsbruglie, 2021-04-20:
@Kronosson Perhaps this issue should be closed?

swamper123, 2021-04-20:
It seems, that this case should have been closed a while ago -- Case closed 🕵️

closed 2021-04-20 by swamper123

Cykorek, 2023-02-18:
`from` opcua import Client
from opcua import ua

url = "opc.tcp://192.168.1.200:4840"
client = Client(url)
client.connect()
print("Client connected")

adres1 = client.get_node("ns=4;i=2")

# Nowa wartość, którą chcemy przypisać do węzła
new_value = ua.Variant(600, ua.VariantType.Int16)

# Tworzymy obiekt DataValue z nową wartością
data_value = ua.DataValue(new_value)

# Przypisujemy nową wartość do węzła
adres1.set_value(data_value)

# Pobieramy wartość węzła, aby upewnić się, że została ustawiona poprawnie
data1 = adres1.get_value()
print(data1)

client.disconnect()`