FreeOpcUa/python-opcua #1160: debug

godassasiom, 2020-11-06:
edp.SecurityPolicyUri = policy.URI

godassasiom, 2020-11-06:
whta‘s problem？

godassasiom, 2020-11-06:
from opcua import ua, Server
server=Server()
server._set_endpoints("opc.tcp://192.168.2.100:48400/freeopcua/server/")
uri = "http://examples.freeopcua.github.io"
idx = server.register_namespace(uri)
server.start()

godassasiom, 2020-11-06:
then debug tell me 
D:\python3.7\python.exe D:/verp_python/opcua_test/opcua_client.py
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
Traceback (most recent call last):
  File "D:/verp_python/opcua_test/opcua_client.py", line 2, in <module>
    opcua-client.exe
NameError: name 'client' is not defined

godassasiom, 2020-11-06:
please help me

AndreasHeine, 2020-11-06:
do not user underscore methods if you dont know what you are doing ( internal methods ) use "server.set_endpoint(" like in the examples 😉

AndreasHeine, 2020-11-06:
Examples:
https://github.com/FreeOpcUa/python-opcua/tree/master/examples

Undescore:
https://www.datacamp.com/community/tutorials/role-underscore-python#SLU
`Single Pre Underscore is used for internal use. Most of us don't use it because of that reason.`