FreeOpcUa/python-opcua #1162: ValueError: 'http://examples.freeopcua.github.io' is not in list

Kronosson, 2020-11-10:
While trying to implement my first subscription i have run into an error. 
Is this only a placeholder for a specific url or is the code just outdated? I wasn't able to find a very recent forum post with this specific problem, so i figured, i should ask here.
I'm using the code shown in the [client example](https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py)

Regards

AndreasHeine, 2020-11-10:
"http://examples.freeopcua.github.io" this is our example namespace if you use one of our example servers there should be a namespace with that url. If you have a different opc ua server the namespace does indeed not exist :) you can use UAExpert to browse the namespace array under the server object to get all avalible namespaces or you can call 
```python
client.get_namespace_array()
```
https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/client/client.py#L581

Kronosson, 2020-11-24:
Alright, i've finally managed to get back to this issue.
As suggested, i'm now using 
`idx = client.get_namespace_array()`

If i run the script at this point, i already get a succesful read of my target node, i want to subscribe to.

Now i'm getting the following issue:

> UaStringParsingError: ("Error parsing string ['http://opcfoundation.org/UA/', 'urn:SIMATIC.S7-1200.OPC-UA.Application:My_PLC_1', 'http://opcfoundation.org/UA/DI/', 'http://www.siemens.com/simatic-s7-opcua', 'http://Server-Interface_1']:MyObject", Value Error ('invalid literal for int() with base 10: [\'http"'))

Am I correct in assuming, this has to do with the `idx = client.get_namespace_array()` ?
Can i manually define the namespace? Right now it's trying to use a string, while it's expecting to find an integer? If i'm able to define the namespace manually, would that help, or just make it more complicated to deal with things at a later date?

swamper123, 2020-11-24:
I think there is still an issue with your code.

I can see that you got all the uris for your namespaces on the server, which you get via `idx = client.get_namespace_array()`.
Now idx is a list of all namespaces you get as return value from the server.

What you want is the index number of an uri: `idx = client.get_namespace_index(uri)`

With that you can go on and browse the server to get your node and continue like in the example code.

Remember! Nodes exist on different namespaces. Each namespace belongs to an uri, which represent a speacial purpose.

AndreasHeine, 2020-11-24:
opc ua addresspace is seperated in namespaces all avalible namespaces of a server are in the NamespaceArray

[image]

### client.get_namespace_array() returns a list of namespaces
### client.get_namespace_index(uri) returns the index in the namespacearray

so if you want to know all namespaces call client.get_namespace_array()
if you know a specific namespacename and you want to access a node you should call client.get_namespace_index(uri) it returns an index and with that index you can ask for a node within this namespace like client.get_node(f"ns={idx};i=100")

long things short 

if you want to build a own adressspace/namespace you need to register it first and then you can fill this namespace with nodes but this is typically serverside code

Kronosson, 2020-11-25:
Thank you both for your help. 
The Problem has now been solved, as i now only ask the server for a specific node, not for a variable node. That solves the need for the namespace index, as i already know it for the two nodes, i want to subscribe to.

swamper123, 2020-11-25:
Nice. :)

Than this case can be closed 🕵️

closed 2020-11-25 by swamper123