FreeOpcUa/python-opcua #1164: How to implement an event from XML information model

ArkasDev, 2020-11-12:
I would like to implement events in OPCUA. For a first POC I created an event programmatically, as shown in the examples. But the standard case is that there is an information model to be implemented. In the example I create an own event object, but how can I apply this procedure to events from the predefined information model?

```
capture_event = None
def create_event(self):
  capture_event_type = self.server.nodes.base_event_type.add_object_type(2, 'CaptureEvent')
  capture_event_type.add_property(2, 'count', ua.Variant(0, ua.VariantType.Double))
  capture_event_type.add_property(2, 'isCaptured', ua.Variant(True, ua.VariantType.Boolean))
  self.capture_event = self.server.get_event_generator(capture_event_type, addressSpaceVisionDetection)

def event_handler(self):
  self.capture_event.Count = self.count
  self.capture_event.trigger(message="Count: %d" % self.count)
```

AndreasHeine, 2020-11-12:
i assume you mean to implement "Alarms & Conditions" part of the OPC UA Spec. described in a Companion Spec. (for VisionSystems)!?
unfortunately "Alarms & Conditions" is not implmeneted in python-opcua, that only means there is no convinient way to use "Alarms & Conditions" so you have to do stuff manually:

implement the ConditionStateMachineType inside the ConditionTypeInstance (the statemachine fires events on statechange)
and the implementation depends on the conditiontype want to implement!

The relevant part in spec.:
https://reference.opcfoundation.org/v104/Core/docs/Part9/
https://reference.opcfoundation.org/v104/Core/docs/Part9/5.1/

ArkasDev, 2020-11-12:
Yes, that would be the long-term goal. However, for this the events defined in the Spec must be implemented. E.g. the `CurrentState` event. As I understood it, maybe I am wrong, I need to load the definition via `self.load_type_definitions("spec.xml")` so that I can access the defined nodes. In the programmatic approach I defined the event type with the properties myself. In the Spec implementation, this step is now already done via the XML definition.

How can I use the event type from the XML definition to create the event on an object, which I loaded via `load_type_definitions`, with the `get_event_generator`?

**CurrentState Event:**
[image]

AndreasHeine, 2020-11-12:
> self.load_type_definitions("spec.xml")

not sure about that!
https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/server/server.py#L513

then (server.load_type...)

https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/server/server.py#L617

then

you can use 
https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/common/node.py#L660

or:
https://github.com/FreeOpcUa/python-opcua/blob/583d970bed30b44236418e03b5a6ff6fea481b13/opcua/common/instantiate.py#L15

and then you need to implement a statemachine logic in in the transitionmethod you need to trigger the event (https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/event_objects.py)

statemachine discribed here (best example i found yet)
https://reference.opcfoundation.org/v104/Core/docs/Part10/5.2.3/#5.2.3.3