FreeOpcUa/python-opcua #1165: Adding methods from Client.

martin4951, 2020-11-16:
I'm running a basic server and adding a method from a client, which seems to add okay, but when calling the method from another client or UAExpert, I get the follow error:

"opcua.ua.uaerrors._auto.BadNothingToDo: "There was nothing to do because the client passed a list of operations with no elements."(BadNothingToDo)"

My client code is:

`import sys
sys.path.insert(0, "..")
import logging
import time

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import Client
from opcua import ua, uamethod

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

@uamethod
def multiply(x, y):
    print("multiply method call with parameters: ", x, y)
    return x * y

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    #logger = logging.getLogger("KeepAlive")
    #logger.setLevel(logging.DEBUG)

    client = Client("opc.tcp://admin@0.0.0.0:4840/freeopcua/server/")
    # client = Client("opc.tcp://admin@localhost:4840/freeopcua/server/") #connect using a user
    try:
        client.connect()
        client.load_type_definitions()  # load definition of server specific structures/extension objects
        uri = "http://examples.freeopcua.github.io"
        idx = client.get_namespace_index(uri)

        
            # get Objects node, this is where we should put our custom stuff
        root = client.get_root_node()
        obj = root.get_child(["0:Objects", "2:Parameters"])

    # populating our address space
        myobj = obj.add_object(idx, "MyObject")
        
        
        inargx = ua.Argument()
        inargx.Name = "x"
        inargx.DataType = ua.NodeId(ua.ObjectIds.Int64)
        inargx.ValueRank = -1
        inargx.ArrayDimensions = []
        inargx.Description = ua.LocalizedText("First number x")
        inargy = ua.Argument()
        inargy.Name = "y"
        inargy.DataType = ua.NodeId(ua.ObjectIds.Int64)
        inargy.ValueRank = -1
        inargy.ArrayDimensions = []
        inargy.Description = ua.LocalizedText("Second number y")
        outarg = ua.Argument()
        outarg.Name = "Result"
        outarg.DataType = ua.NodeId(ua.ObjectIds.Int64)
        outarg.ValueRank = -1
        outarg.ArrayDimensions = []
        outarg.Description = ua.LocalizedText("Multiplication result")
 
        multiply_node = obj.add_method(idx, "multiply", multiply, [inargx, inargy], [outarg])
    
        root = client.get_root_node()
        print("Root node is: ", root)
        objects = client.get_objects_node()
        print("Objects node is: ", objects)

        # Node objects have methods to read and write node attributes as well as browse or populate address space
        print("Children of root are: ", root.get_children())

        # gettting our namespace idx
        uri = "http://examples.freeopcua.github.io"
        idx = client.get_namespace_index(uri)

        # Now getting a variable node using its browse path

        
        

        # calling a method on server
        res = obj.call_method("{}:multiply".format(idx), 3, 9)
        #print("method result is: ", res)

        embed()
    finally:
        client.disconnect()`

Is it not possible to add methods from a client?

AndreasHeine, 2020-11-16:
1. only servers can have methods
2 clients can add/delete nodes 

so you can add the method node but the server needs to know the callbackfunction in your code only the client knows it but the server does not!

martin4951, 2020-11-16:
Hi, thanks for your response. So it would be correct to say that methods can never work with client data, as the client is added after the server is ran (and methods/callbacks are already defined)? So instead of using methods to query the status of a client node (from one client to another), you would have to use subscription based data change notifications?

AndreasHeine, 2020-11-16:
yes! subscription and the eventstuff can handle that. uamethods are just some predefined funtionality which the client can invoke if he wants to.

martin4951, 2020-11-16:
Great, thanks for your help 👍

closed 2020-11-19 by martin4951