FreeOpcUa/python-opcua #1166: No Extension class registered while loading PackML NodeSet

pbertoni89, 2020-11-17:
I need to implement a PackML server (with additional custom objects btw). So since it seems missing from FreeOpcUa repos, I'm referring to [UA-Nodeset](https://github.com/OPCFoundation/UA-Nodeset/) repo.

Within that context, I'd import inside the Modeler the following [Opc.Ua.PackML.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/PackML/Opc.Ua.PackML.NodeSet2.xml), but it fails with

```
uamodeler.model_manager - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
uamodeler.server_manager - INFO - Starting python-opcua server')
opcua.server.server - WARNING - Endpoints other than open requested but private key and certificate are not set.')
opcua.server.binary_server_asyncio - WARNING - Listening on 0.0.0.0:48400')
opcua.common.xmlimporter - WARNING - failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=6018))')
uawidgets.utils - ERROR - ('Error no extension class registered ', 'PackMLVersionDataType', NumericNodeId(ns=1;i=13))')
Traceback (most recent call last):
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/uamodeler.py", line 249, in open
    self._model_mgr.open(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/model_manager.py", line 181, in open
    self.open_xml(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/model_manager.py", line 113, in open_xml
    self._open_xml(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/model_manager.py", line 119, in _open_xml
    path = self.import_xml(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/model_manager.py", line 102, in import_xml
    new_nodes = self.server_mgr.import_xml(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/server_manager.py", line 68, in import_xml
    return self._backend.import_xml(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/uamodeler/server_manager.py", line 112, in import_xml
    return self._server.import_xml(path)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/server/server.py", line 510, in import_xml
    return importer.import_xml(path, xmlstring)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 67, in import_xml
    node = self._add_node_data(nodedata)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 86, in _add_node_data
    node = self.add_variable(nodedata)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 207, in add_variable
    attrs.Value = self._add_variable_value(obj,)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 305, in _add_variable_value
    extobj = self._make_ext_obj(obj.value)
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 238, in _make_ext_obj
    ext = self._get_ext_class(obj.objname)()
  File "/media/xnext/DATAEXT/opt/miniconda/envs/ml36/lib/python3.6/site-packages/opcua/common/xmlimporter.py", line 233, in _get_ext_class
    raise Exception("Error no extension class registered ", name, nodeid)
Exception: ('Error no extension class registered ', 'PackMLVersionDataType', NumericNodeId(ns=1;i=13))
```

Instead, the [Opc.Ua.Di.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/DI/Opc.Ua.Di.NodeSet2.xml) of the same repo seems to work well, just logging three lines:

> opcua.common.xmlparser - WARNING - Parsing value of type 'QualifiedName' not implemented')
> uamodeler.model_manager - WARNING - Could not find struct TransferResultErrorDataType under i=22')
> uamodeler.model_manager - WARNING - Could not find struct TransferResultDataDataType under i=22')

Are extension classs strictly needed? I guess I cannot modify the PackML NodeSet while pretending to be compliant to that companion. What should I do in order to import PackML, being able to extend it afterwards?

zerox1212, 2020-11-17:
Maybe the import order is not correct. It's trying to add an extension object that doesn't exist yet. If it were me, I would run modeler from an IDE and import small parts of the XML and try to troubleshoot why it isn't happy. Good luck, the XML stuff is a bit complicated.

swamper123, 2020-11-18:
There are known issues about the xml import, sometimes missing Nodes and References. Perhaps this is happening in your case as well. And like @zerox1212 said, it will need time to have a fix for that, but I try my best to get stuff running.

pbertoni89, 2020-12-17:
> There are known issues about the xml import, sometimes missing Nodes and References. Perhaps this is happening in your case as well. And like @zerox1212 said, it will need time to have a fix for that, but I try my best to get stuff running.

Same problem found on asyncua v0.9.3. PackML Nodeset copied from [v1.04 - 1abd22f](https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04). Since I'm quite new to OPCUA, I'd be glad to get few hints on what could I write by myself. 

What exactly is an "extension class" ? Why it's missing from [that particular xml file](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/PackML/Opc.Ua.PackML.NodeSet2.xml) ?

pbertoni89, 2020-12-21:
Issue solved with latest UA-Nodeset updates (perhaps [2afcf77](https://github.com/OPCFoundation/UA-Nodeset/commit/2afcf77c6d889701c040c09379ac06ec2febacc8))

closed 2020-12-21 by pbertoni89