FreeOpcUa/python-opcua #1168: ERROR CREATING SUBSCRIPTION

gabino-alonso, 2020-11-19:
Hello all, I am have a little problem when I try to create a subscription with my server (Omron NX PLC)

when I test this code:

```
client.connect()
root = client.get_root_node()
print("Root is", root)
print("childs of root are: ", root.get_children())
print("name of root is", root.get_browse_name())
objects = client.get_objects_node()
print("childs og objects are: ", objects.get_children())

 tag1 = client.get_node("ns=4;s=DeviceStatus.ErrorStatus")
print("tag1 is: {0} with value {1} ".format(tag1, tag1.get_value())) 
tag2 = client.get_node("ns=4;s=DeviceStatus.Mode")
print("tag2 is: {0} with value {1} ".format(tag2, tag2.get_value()))
```
so, the result was fine (atached below):
```
Root is i=84
childs of root are:  [Node(TwoByteNodeId(i=87)), Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86))]
name of root is QualifiedName(0:Root)
childs og objects are:  [Node(FourByteNodeId(i=2253)), Node(FourByteNodeId(ns=2;i=5001)), Node(StringNodeId(ns=4;s=NxController))]
tag1 is: ns=4;s=DeviceStatus.ErrorStatus with value NoError 
tag2 is: ns=4;s=DeviceStatus.Mode with value RUN 
```
but when add the instructions to create a subscription, unfortunately I have an error ..
New code:
```
handler = SubHandler()
sub = client.create_subscription(0, handler)
```
 the error:
```
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
ERROR:concurrent.futures:exception calling callback for <Future at 0x7f71b40c4ac0 state=finished returned Buffer>
```
Please, could someone help me ?
Thanks a lot ...

AndreasHeine, 2020-11-19:
you should not use a subscription period of 0, everything below 100ms is unrealistic in plc check the docs or settings from the plc what the smalest publish interval is the plc can provide!

swamper123, 2020-11-19:
1) The only reason to write Capslock is, that your Capslock key is jammed. But normally you should fix that first and then ask questions. Nobody would help you faster if it is written in Capslock (sometimes people won't respond to it, because of that, like I do on other platforms). If you want help by a problem, type the issue like everybody else and be patient (especially in OpenSource Projects). For the next time. Anyway!
2) Why did you set the RequestedPublishingInterval to 0?
3) I guess you added the SubclassHandler in your code as well, like in the examples shown?

gabino-alonso, 2020-11-19:
> you should not use a subscription period of 0, everything below 100ms is unrealistic in plc check the docs or settings from the plc what the smalest publish interval is the plc can provide!

thanks for you response,  if I change the value for 500ms the result is the same.

gabino-alonso, 2020-11-19:
> 1. The only reason to write Capslock is, that your Capslock key is jammed. But normally you should fix that first and then ask questions. Nobody would help you faster if it is written in Capslock [...]

OK, sorry ... I know for next time ...

>     2. Why did you set the RequestedPublishingInterval to 0? (See description why at @AndreasHeine comment, who is always a step ahead of me stuck_out_tongue_closed_eyes  )
> 

was a mistake, now is in 500ms and the result is the same ...

>     3. I guess you added the SubclassHandler in your code as well, like in the examples shown?

this ?

```
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)
```

thanks again for your help ...

swamper123, 2020-11-19:
@gabino-alonso It's alright. 😉 

Did you even fetch one Subscription? They only wouldn't respond either if they are on their limit (because the Prio-Job of a PLC is to do the PLC programmed stuff). Did you tried something like 2000ms, which should be realy enough time?

And yes that SubHandler class, but I guess you already had it.

Since I don't know the Nx Series well, can you post the exaxt device name (or even better a link to the docs)?

gabino-alonso, 2020-11-20:
> Did you even fetch one Subscription? They only wouldn't respond either if they are on their limit (because the Prio-Job of a PLC is to do the PLC programmed stuff). Did you tried something like 2000ms, which should be realy enough time?
<

No, I have not had an answer never, only that error when I create the subscription. I have changed the timeout by 2000 with the same result.

>Since I don't know the Nx Series well, can you post the exaxt device name (or even better a link to the docs)? Is there a minimum responds time (or similar) mentioned for responstimes or subscriptions?
<
My Omron PLC is a NX102 9000 and is a PLC only for testing and is not in a work environment and its workload is only of the 1% 

In the manual (attached) does not have any information about response time, only about number of subscriptions, vars, etc ...

thanks again ...

[w588_nj_nx-series_opc_ua_cpu_units_users_manual_en.pdf](https://github.com/FreeOpcUa/python-opcua/files/5573481/w588_nj_nx-series_opc_ua_cpu_units_users_manual_en.pdf)

swamper123, 2020-11-20:
Okay, suspicious, but keeps us a step further to a solution.
 
Have you tried another Client? Like [UAExpert](https://www.unified-automation.com/products/development-tools/uaexpert.html)?
(Connect to your Server, go do Document->Add-> Datalogger View which is the same as the subscription (on value changes) -> Drag&Drop the Node you desire into the window and make the necessarry configs)

If that one works, we may have to go deeper in the bit & bytes.

Alternative, does the [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio) work? 
The Code is pretty much the same for the Client. (If that works, we know it's beeing around here in the repo).

gabino-alonso, 2020-11-20:
With UAExpert works fine (I think ...) attached the Data Logger View screen and the result file:

``
PrimaryKey,DataType,Value,StatusCode,SourceTimeStamp,ServerTimeStamp
1,0x0006,999,0x00000000,2020-11-20T05:24:12.908Z,2020-11-20T05:24:12.908Z
2,0x0001,false,0x00000000,2020-11-20T05:24:12.908Z,2020-11-20T05:24:12.908Z
3,0x0005,1,0x00000000,2020-11-20T05:24:12.908Z,2020-11-20T05:24:12.908Z
4,0x0001,false,0x00000000,2020-11-20T05:24:12.908Z,2020-11-20T05:24:12.908Z
5,0x0001,false,0x00000000,2020-11-20T05:24:12.908Z,2020-11-20T05:24:12.908Z
1,0x0006,1000,0x00000000,2020-11-20T05:24:50.903Z,2020-11-20T05:24:50.903Z
2,0x0001,true,0x00000000,2020-11-20T05:24:50.903Z,2020-11-20T05:24:50.903Z
2,0x0001,false,0x00000000,2020-11-20T05:24:55.238Z,2020-11-20T05:24:55.238Z
1,0x0006,1001,0x00000000,2020-11-20T05:24:57.788Z,2020-11-20T05:24:57.788Z
2,0x0001,true,0x00000000,2020-11-20T05:24:57.788Z,2020-11-20T05:24:57.788Z
2,0x0001,false,0x00000000,2020-11-20T05:24:59.318Z,2020-11-20T05:24:59.318Z
1,0x0006,2222,0x00000000,2020-11-20T05:25:15.128Z,2020-11-20T05:25:15.128Z
``

[image]

Now I am goin to test with the Async library.

For me it is indifferent to work with one or the other version since it is a new project in the medium term.
We are preparing a SCADA to visualize production data from the line PLCs using Python + SQL + Dash.
Which version would be better to use or is it indifferent?

thanks

AndreasHeine, 2020-11-21:
> Which version would be better to use or is it indifferent?

asyncua is better for clients!

swamper123, 2020-11-23:
Sound like that this case can be closed. 🕵️

closed 2020-11-23 by swamper123