FreeOpcUa/python-opcua #1169: LDS, GDS implementation

RakshanPremsagar, 2020-11-20:
Hello,
PS: Just a beginner doing a literature research as to the features supported by PythonOpcua.

From looking at the client source file, there have been implementations for OPC UA discovery services.
But there is no Python based implementation of LDS / GDS / LDS -ME?

Would it be possible to use the LDS specified by the OPC Foundation and use the Python stack to register the server to the LDS?
And what about the GDS implementation?

oroulet, 2020-11-20:
there is some code for it. But I am not sure it is much tested... probably some bugs

swamper123, 2020-11-20:
Just recognized that opcua-asyncio hasn't got the LDS class, which exist here. Is that a leftover?
Since there aren't any tests for `LocalDiscoveryService` you shouldn't count on that and I can't see that other discovery services are covered well. 
I also don't see that this feature will be implemented & tested in near future (neither in [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio), which is the sequel repo of python-opcua).

AndreasHeine, 2020-11-23:
looks like LDS-ME is not Implemented!