FreeOpcUa/python-opcua #1171: Adding Engineering Unit in python opcua

ksrangini05, 2020-11-23:
Hi,
How to add an engineering unit to a variable as per OPC UA specification? I want it to reflect in UA expert tool . From OPC UA schemas,  I have seen that there is a csv file mentioning units  - UNECE code, unit ID,etc. Whether I have to use it.?

AndreasHeine, 2020-11-23:
EngeneeringUnit has type EUInformation !?

you can find it under Types -> DataTypes -> BaseDataTypes -> Structure -> EUInformation ("ns=0;i=887")
and the python classes are part of "ua.uaprotocol_auto" (you need "from opcua import ua")

https://github.com/FreeOpcUa/python-opcua/blob/278a2b3673736d7aa9f00799c86d79e43d50bd5b/opcua/ua/uaprotocol_auto.py#L10996

so its a structure / "extensionobject"

```python
    # create a variable (EngineeringUnits) of type EUInformation ("ns=0;i=887") 
    units = ua.uaprotocol_auto.EUInformation()
    """
    write to the fields according to:
    https://reference.opcfoundation.org/v104/Core/docs/Part8/5.6.3/

    :ivar NamespaceUri:
    :vartype NamespaceUri: String
    :ivar UnitId:
    :vartype UnitId: Int32
    :ivar DisplayName:
    :vartype DisplayName: LocalizedText
    :ivar Description:
    :vartype Description: LocalizedText
    """
    EngineeringUnits.set_value(ua.Variant(units , ua.VariantType.ExtensionObject))
```

swamper123, 2020-11-23:
> EngeneeringUnit has type EUInformation !?

Yep. [See Specs.](https://reference.opcfoundation.org/v104/Core/docs/Part8/5.6.3/)
Probably you have to parse the csv file (via hand (strg+f) or via python ) for your wanted Unit.

NamespaceUri should be  http://www.opcfoundation.org/UA/units/un/cefact except you have another source.
The unitId is listed in the csv file (I don't know how they create that one) with the other attributes (DisplayName, Description).

AndreasHeine, 2020-11-23:
@swamper123 see my comment ;)

> """
>     write to the fields according to:
>     https://reference.opcfoundation.org/v104/Core/docs/Part8/5.6.3/
>
>     :ivar NamespaceUri:
>     :vartype NamespaceUri: String
>     :ivar UnitId: [...]

swamper123, 2020-11-23:
@AndreasHeine Your comment sounded unsure, so I wanted to make that clear. :^)

AndreasHeine, 2020-11-23:
ok now you confused me -_-°

ksrangini05, 2020-11-24:
> EngeneeringUnit has type EUInformation !?
>
> you can find it under Types -> DataTypes -> BaseDataTypes -> Structure -> EUInformation ("ns=0;i=887")
> and the python classes are part of [...]

Where should I send the Unit ID ?  Here it says int 32!

1. first I  am trying to have the engineering units column in expert tool with a test code

from opcua import ua,  Server
from opcua.common.ua_utils import get_nodes_of_namespace
import time
from random import randint

server = Server()

server.set_endpoint("opc.tcp://localhost:4840/freeopcua/server/")

name = "OPC UA Simulation server"

add_space = server.register_namespace(name)

node = server.get_objects_node()
param = node.add_object(add_space,"Parameters")
temp = param.add_variable(add_space, "Temperature",0)
temp.set_writable()
server.start()
print("server started at ")

while True:
    temperature = randint(10,500)
    temp.set_value(temperature)

    time.sleep(5)

2. Once I understood, I need to implement for n number of data points specified in json file (to get value for each datapoint from a influx db)

swamper123, 2020-11-24:
> Where should I send the Unit ID ? Here it says int 32!

? 

From the file  UNECE_to_OPCUA you add this number for your Unit. This is an int32 number.
[image]

Or did you mean the Value?

ksrangini05, 2020-11-27:
@AndreasHeine 
Hi, Is it possible to add engineering units as a column in UA expert tool like "NodeId", "Value", etc?  Do ypu have any examples for this? Still I am trying on this. !
Thanks

AndreasHeine, 2020-11-27:
@ksrangini05 i am not sure what you mean (apologize for that) i need more information what exacly you mean!?

with collumn do you mean the node-attribute-tab in uaexpert?

this is an example how a analogitem would look like:
so you would have a variable with the numeric value and under it the propertys with the information about it like range "0-1000" and for example as pysical unit "cm"
[image]
[image]
[image]

ksrangini05, 2020-11-27:
[image]

ksrangini05, 2020-11-27:
I am having different parameters like voltage(Voltage, kilo-volt, mega-volt), current, frequency(Hz), power (kW, MW, GW), energy (kWh,GWh,MWh),etc., in one window. Hence I need to show all different parameters with different engineering units in UA tool. In the below picture , you can see different parameters which are different and having different units. 

[image]

AndreasHeine, 2020-11-27:
this is not a python-opcua topic or its better to say you cant add them via opcua the thing is the temperatur node only contains the value itself and not its unit even if you use a analogitemtype the units would be in another row (its another datapoint!)

AndreasHeine, 2020-11-27:
if you want to look on a sample server:
opc.tcp://uademo.prosysopc.com:53530/OPCUA/SimulationServer

ksrangini05, 2020-11-29:
Thanks - I have had some success with defining an EUInformation node by using the below:

**In the section where I am populating the address space**

myprop = myvar.add_property(_namespace_, _bname_,0,None,ua.object_ids.ObjectIds.EUInformation)

However, I am struggling to assign EUInformation to the above node

units = ua.uaprotocol_auto.EUInformation()
                                units.NamespaceUri= _namespace_
                                units.UnitId = 4536630
                                units.DisplayName = "°/s"
                                units.Description = "degree per second"

_propertynode_.set_value(ua.Variant(units))

When I tried the above to set units, I am getting this error - **AttributeError: 'str' object has no attribute 'ua_types'**

While I can set a normal integer value to this property by using the below,

_propertynode_.set_value(12345)

 I am unable to assign EUInformation value to the same. Please help.

AndreasHeine, 2020-11-29:
`propertynode.set_value(ua.Variant(units, ua.VariantType.ExtensionObject))`

and you need to assign the correct datatypes...

units.DisplayName = "°/s" -> ua.LocalizedText("°/s", "en-US")
units.Description = "degree per second" -> ua.LocalizedText("degree per second", "en-US")

ksrangini05, 2020-11-29:
> `propertynode.set_value(ua.Variant(units, ua.VariantType.ExtensionObject))`
>
> and you need to assign the correct datatypes...
>
> https://reference.opcfoundation.org/v104/Core/docs/Part8/5.6.3/ [...]

thank you..it works but without "en-US"

AndreasHeine, 2020-11-29:
> it works but without "en-US"

thanks for the hint there is a missing arg. in class init

AndreasHeine, 2020-11-30:
@ksrangini05 in the next release is it possible to pass the locale like `ua.LocalizedText("degree per second", "en-US")`