FreeOpcUa/python-opcua #1172: Node history from client.

martin4951, 2020-11-23:
Is it possible to historize a node that has been generated by a client? I know this is possible through the server using 

`history_manager`
and
`historize_node_data_change`

But there doesn't seem to be any methods available from client generated nodes.

AndreasHeine, 2020-11-23:
oh thats not a typical usecase! i would use a uamethod to create the node and add the historizing logic from that methodcallback because if the client adds the node somewhere in the addressspace, the server does not know he is there and off course not what to do with it, its a node without serverlogic. with a uamethod you can explicitly say create a node under this parent with this type and with historizing ...

AndreasHeine, 2020-11-23:
this part of the opcua reference might be interesting for you https://reference.opcfoundation.org/v104/Core/docs/Part4/5.7.2/