FreeOpcUa/python-opcua #1173: Regarding Not able to Historical Data From WinCC SCADA for tags of TEXT TYPE

rajnishnationfirst, 2020-11-24:
Hi ,

We need your assistance as we are stuck at a very important stage and we are not able to proceed further.

We are facing issue for a situation when fetching data . Below is the issue described.

1)	There are 15-20 tags of type Text at WinCC Side for holding values of type text – having combination of numbers and letters.
2)	We are able to fetch the data for other tags of type floating point but we are not getting data ( We are not getting any error and we are getting empty array) from OPCUA client for the tags of TYPE text over OPCUA HDA. Also we crossed check the issue in UA Expert , but here also we are not able to see any data .
3)	Also in WinCC its showing the data is being logged in its consle interface but for both clients Python Code Client and UA Expert Client are not showing any data .

As per Python OPCUA  documentation .. node.read_raw_history(start_time, end_time , 0 ) works on nodes of any data type ( where node type should be variable )

Pls see the below screenshots for your reference :

1)
**WinCC SCADA Snap: ( WinCc SCADA snap , its shows Data )**

[image]

2)

Python OPCUA ( free OPCUA ) Snap ( There is not data fetched ) - getting empty array

[image]

3) UA Expert Client : ( Here also i am getting emplty values ) 

[image]

What may be the issue , pls assist .. i am not able to track .. whether its SCADA end issue or the PYHTON OPCUA client & UA Expert issue.

Thanks 

Rajnis Vishwakarma

zerox1212, 2020-11-24:
If UA Expert can't read it, then I highly doubt it's a problem on the read side. I wouldn't even bother using Python OPC UA until you have it working in UA Expert.

rajnishnationfirst, 2020-11-24:
Thank You Andrew. 

It would be  great of you if you can help or tell on how we can do the following below :

1) What is the correct configuration of the nodes of NODE TYPE Variable of DATA Type TEXT,  String,  Character, etc any other data type holding values having TEXT both as combination of characters and Numbers like ' MT123456 ' or having alone only numbers or alone having characters, string? 

2) I know that node.read_raw_history(starttime , endtime, 0) is used for historical data for all nodes of NODE TYPE VARIABLE of any DATA TYPE as per OPCUA Python Documentation,  will I will be able to fetch all historical data having combination of characters and Numbers using the same node.read_raw_history(starttime , endtime, 0) ? 

3) Which data type will be more preferable in the SCADA ( Siemens WinCC Scada) to hold combination of characters + numbers like
 ' MT123456 ' or individually alone number or only characters? 

Andrew pls assist on the above points as I m not able to fetch the data for the DATA TYPE having combination of characters + numbers using the nodes function for historical access node.read_raw_history(starttime , endtime, 0) ?

Thanks and Regards 

Rajnish Vishwakarma

zerox1212, 2020-11-25:
Sorry I don't use Siemens products. If UA Expert can't read your WinCC strings then I would probably contact Siemens for support. I do not think volunteers on an open source Python library is the best place to solve this problem.

duduyoyo, 2024-02-01:
There is a HDA solution for Python [here](https://github.com/duduyoyo/WebSocket4OPC). Good luck!