FreeOpcUa/python-opcua #1174: Regarding getting error when Historical Alarms and Events fetched through python opcua's  read_event_history(starttime=None, endtime=None, numvalues=0, evtypes=2041)

rajnishnationfirst, 2020-11-25:
Hi Colleagues,

I am fetching **historical  alarms and events** over **OPCUA Protocol** .

Alarms and Events are logged at the SCADA side ( Siemens WinCC SCADA ) and is visible at WinCC Side .

Also i am getting alarms and events in real time from handler's using teh below code suceesfully over OPCUA:

**class SubHandler(object):

def datachange_notification(self, node, val, data):
    pass
    #print("Python: New data change event", node, val)

def event_notification(self, event):
    print("Python: New event", event)**

Below are the points which  i need help to fecth alarms and events from historical alarms and events ove r OPCUA: 

1) Historical alarms are fetched using node function **node.read_event_history(starttime=None, endtime=None, numvalues=0, evtypes=2041)** ,  

1) Will opcUA **node.read_event_history(starttime=None, endtime=None, numvalues=0, evtypes=2041)** has different nodes other than data nodes ( Means are Alarms nodes are different and data nodes are different ? ) ?

2) **node.read_event_history(starttime=None, endtime=None, numvalues=0, evtypes=2041)** when called on alarms and events node id's i a getting error as below .

[image]

3) When i m calling the node.read_raw_history(starttime=None, endtime=None, numvalues=0, evtypes=2041) on alarms nodes i am getting Historical Data's but not historical alarms and events .

Some questions : 

1) Can i use **node.read_event_history(starttime=None, endtime=None, numvalues=0, evtypes=2041)** on OPCUA server  for fetching historical alarms and events ?

2) Or is there is need to configure any other server's like OPC A&E Server separately to fecth historical alarms and events ?

Would be great if anyone from the group can assist me on this.

Thanks and Regards
Rajnish Vishwakarma

AndreasHeine, 2020-11-25:
does "BadNotImplemented" came from WinCC? look in the wiresharke trace! what is the response on "read_event_history()"

rajnishnationfirst, 2020-11-25:
Andrew,

Thanks for the support and help you are giving me these days and its very great of you giving time to solve the issues.

Yes when i used **node.read_event_history(starttime=None, endtime=None, numvalues=0, evtypes=2041)**  on alarms nodes ( TAGS ) and queried WinCC SCADA over OPCUA server , i got the error / exception as "BadNotImplemented" ( Pls check the above snap attahed above in the post.)

Thanks and Regards
Rajnish Vishwakarma

AndreasHeine, 2020-11-25:
the name is Andreas but i dont mind ;)

> ( Pls check the above snap attahed above in the post.)
i did! but again if its a response from WinCC OPC UA Server which says "BadNotImplemented" means that the OPC UA Server does not Support what you asked him for!

what WinCC is it there are a lot... tbh!