FreeOpcUa/python-opcua #1181: trying to get the children of a node that has lot's of children: Not enough data left in buffer

juamar, 2020-12-01:
Hi,
I 'm trying to get the children of a node that has lot's of children, like this:

node.get_children(refs=33)

I don't know how to determine the number of children it has, but is pretty big. UA Experts takes some minutes to bring all the nodes (which are signals with values) The problem is that when my codes tries to execute that line, i get this Exception:

Not enough data left in buffer, request for 960036864, we have Buffer(size:6523938, data:b'999|PT9801\x03\x05\x.... and it starts printing all the data in the buffer... i suppose.

I have tried to use max_messagesize parameter to see if this increases this buffer, but it's the same.

Is there something i can do to handle this? Maybe there is a way to grt the children in a way that doesn't need to retrieve all the data at once.

BR!

AndreasHeine, 2020-12-01:
could you provide some information about the server (vendor, stack, language) and about the variable type

juamar, 2020-12-01:
Hi,
I only have the configuration name right now:
open62541-based OPC UA Application
As far as ir remember, this publishes some tags in a SCADA System.

About the Node which is givin problems i have this...
The nodeClass is an object. It contains tags, like floats, and ints, and so on.
It has type Definition as "StationType", Property "StationNumber", and Components... like this:

[image]

The components in this case will be the signals.
Is this information usefull?

BR!

AndreasHeine, 2020-12-02:
could you turn on the logger?

```python
import logging
logging.basicConfig(level=logging.DEBUG)
```

and post the browse console output!
a stacktrace would also be helpfull

juamar, 2020-12-02:
Okey, for the sake of simplicity i did this code to specially test this situation:

<pre>
import logging
from opcua import Client
from NodeXmlExporter import NodeXMLExporter

logging.basicConfig(filename="trace.log",level=logging.DEBUG)

exporter = NodeXMLExporter()
exporter.connect("opc.tcp://50.200.105.156:4840")
nodeid="ns=1;s=99999|DASMAP"
node = exporter.client.get_node(nodeid)
node.get_children(refs=33)
exporter.client.disconnect()
</pre>

Now, the exporter connect goes like this:

<pre>
import argparse
import logging
import sys
import time
from opcua import Client
from XmlExporter import XmlExporter

class NodeXMLExporter:
    def __init__(self):
        self.nodes = []
        self.namespaces = {}
        self.visited = []
        self.client = None
        self.logger = logging.getLogger(__name__)
        #self.logger.setLevel(logging.INFO)

    ...

    def connect(self, server_url="opc.tcp://localhost:16664"):
        from opcua.crypto import security_policies
        import types
        from opcua.ua.uaprotocol_hand import CryptographyNone
        
        self.client = Client(server_url)
        self.client.session_timeout = 3600000
        
        # Fix symmetric_key_size (not 0) of securityPolicy
        sec_policy = security_policies.SecurityPolicy()
        sec_policy.symmetric_key_size = 8
        self.client.security_policy = sec_policy
        
        # Fix signature method of CryptographyNone
        def signature(self, data):
            return None
        fixed_signature = types.MethodType(signature, CryptographyNone)
        self.client.security_policy.asymmetric_cryptography.signature = fixed_signature
        self.client.max_messagesize = 99999999
        
        try:
            self.client.connect()
            self.client.max_messagesize = 99999999
            self.client.session_timeout = 3600000
        except Exception as e:
            self.logger.error("No connection established", e)
            self.logger.error(e)
            self.logger.error("Exiting ...")
            sys.exit()

        self.logger.info("Client connected to %s" % server_url)
        self.logger.debug("Timeout = %d" % self.client.session_timeout)
[... 234 lines cut ...]
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'C', body_size:65522, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:b'MSG', chunk_type:b'F', body_size:38447, channel:3)
DEBUG:opcua.uaprotocol:Waiting for header
</pre>

And the the program starts printing the buffer error i mention in the beggining, which is so big it overflows the console buffer, and i cannot get it as a hole unless i do i try catch of the error and tell the code to print it in some file, as i was doing earlier with the logging library. As i said before, the error goes like this:

> Not enough data left in buffer, request for 960036864, we have Buffer(size:6523938, data:b'999|PT9801\x03\x05\x.... and it starts printing all the data in the buffer... i suppose.

I saw yesterday that raise that prints this error is located in a Buffer class within your library.

Thanks for your help!

AndreasHeine, 2020-12-02:
ok

max_messagesize = 99999999 -> the real max messagesize is negotiated between client and server during the Hello/Ack handshake on connect 
https://github.com/FreeOpcUa/python-opcua/blob/131ad8ceffeb7510b7bb3d9db1742896dbf74196/opcua/client/ua_client.py#L180

means if the server does not support "99999999" the actually used max messagesize might be lower! if you want to check that you can use wireshark and trace the connect filter "opcua" and you can check "hello" and the response "ack" the smaller messagesize will be used during the connection

so the main question is now what is the used messagesize?

juamar, 2020-12-02:
Okeeyy, interesting.

This is the Hello:
[image]

And this the acknowledge:
[image]

I think maybe the problem is not in the Message Size, but in the buffer size... but the error says the buffer size is different.

juamar, 2020-12-02:
I'm trying to analize the response for the browse the code is doing internally. this is the result:

[image]

But the code is now telling me this (only on console):

--- Logging error ---
Traceback (most recent call last):
  File "d:\javendano\windows\Codes\opc_ua_xml_export_client\NodeXmlExporter.py", line 143, in connect
    self.client.connect()
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\client\client.py", line 272, in connect
    self.connect_socket()
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\client\client.py", line 307, in connect_socket
    self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\client\ua_client.py", line 266, in connect_socket
    return self._uasocket.connect_socket(host, port)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\client\ua_client.py", line 155, in connect_socket
    sock = socket.create_connection((host, port), timeout=self.timeout)
  File "C:\Program Files\Python39\lib\socket.py", line 843, in create_connection
    raise err
  File "C:\Program Files\Python39\lib\socket.py", line 831, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\logging\__init__.py", line 1079, in emit
    msg = self.format(record)
  File "C:\Program Files\Python39\lib\logging\__init__.py", line 923, in format
    return fmt.format(record)
    record.message = record.getMessage()
  File "C:\Program Files\Python39\lib\logging\__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "d:\javendano\windows\Codes\opc_ua_xml_export_client\opcuaTeamHelp.py", line 8, in <module>
    exporter.connect("opc.tcp://localhost:4840")
  File "d:\javendano\windows\Codes\opc_ua_xml_export_client\NodeXmlExporter.py", line 147, in connect
    self.logger.error("No connection established", e)
Message: 'No connection established'
Arguments: (ConnectionRefusedError(10061, 'No connection could be made because the target machine actively refused it', None, 10061, None),)
PS D:\javendano\windows\Codes\opc_ua_xml_export_client> & "C:/Program Files/Python39/python.exe" d:/javendano/windows/Codes/opc_ua_xml_export_client/opcuaTeamHelp.py
Traceback (most recent call last):
  File "d:\javendano\windows\Codes\opc_ua_xml_export_client\opcuaTeamHelp.py", line 11, in <module>
    node.get_children(refs=33)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\common\node.py", line 316, in get_children
    return self.get_referenced_nodes(refs, ua.BrowseDirection.Forward, nodeclassmask)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\common\node.py", line 390, in get_referenced_nodes
    references = self.get_references(refs, direction, nodeclassmask, includesubtypes)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\common\node.py", line 369, in get_references
    results = self.server.browse(params)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\client\ua_client.py", line 327, in browse
    data = self._uasocket.send_request(request)
  File "C:\Users\javendano\AppData\Roaming\Python\Python39\site-packages\opcua\client\ua_client.py", line 83, in send_request
    data = future.result(self.timeout)
  File "C:\Program Files\Python39\lib\concurrent\futures\_base.py", line 442, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

Previous executions where made on linux, but this one is being made on windows.
Looks like the error is the same, but on windows Python will return timeout, and on linux the real error.

juamar, 2020-12-02:
Some update. If i set self.client.secure_channel_timeout = 3600000 as a connection parameter, the code will return the same as error as on linux. Messages on Wireshark are exactly the same.

juamar, 2020-12-02:
Also, it seems like UAExpert uses BrowseNext for the task instead of Browse. UAExpert is able to bring all the references to the components of the Node.

AndreasHeine, 2020-12-02:
get_children uses browse_next aswell!
https://github.com/FreeOpcUa/python-opcua/blob/131ad8ceffeb7510b7bb3d9db1742896dbf74196/opcua/common/node.py#L294
https://github.com/FreeOpcUa/python-opcua/blob/131ad8ceffeb7510b7bb3d9db1742896dbf74196/opcua/common/node.py#L384
https://github.com/FreeOpcUa/python-opcua/blob/131ad8ceffeb7510b7bb3d9db1742896dbf74196/opcua/common/node.py#L348

https://github.com/FreeOpcUa/python-opcua/blob/131ad8ceffeb7510b7bb3d9db1742896dbf74196/opcua/client/ua_client.py#L333

AndreasHeine, 2020-12-02:
could you post the server capabilities?

like this:
[image]

juamar, 2020-12-02:
I don't understand much about OPCUA and the protocol, but the Python is doing the request like this:

[image]

And OPC UA something like this (much longer i have to say):

[image]

These are the server capabilities:

[image]

AndreasHeine, 2020-12-02:
::1 = IPv6 Localhost
127.0.0.1 = IPv4 Localhost

juamar, 2020-12-02:
With IP4 is the same result as with IP6, and it executes the same way.

zerox1212, 2020-12-02:
Can you change the server? Is 8mb of children really a smart design?

starturtle, 2020-12-02:
Does the parent node have huge children, or are there indeed very many? How about limiting the max number of references to return? If I remember correctly, that part can be tweaked on both sides. In that case it should be possible to stream and process the result list in chunks.

AndreasHeine, 2020-12-02:
thats actually browse next does! but i am not sure if 8mb is possible with <8mb buffer 🤔 because i cant construct the message in one shot

juamar, 2020-12-02:
@starturtle There are very many. The children are just variables.

juamar, 2020-12-02:
I am encountering the same problem using another library. As i try to find a solution, i will see if there is a chance of dividing that node in many nodes with less children (or group by inner nodes).

AndreasHeine, 2020-12-02:
or you need to configure the server with a bigger buffer! what  in my opinion is a serverside error is that the server does not return a statuscode he just sends not all data... the buffer should always be as big as the biggest message it can send!

juamar, 2020-12-04:
Finally i am using a .NET library for the pourpose we where trying to achieve. The solution would be to limit in some way how many nodes we retrieve per request. I insist, the solution i applied was to do first a browse request with, ie 100 nodes, and then the subsequent request to do it with browseNext. When i do <pre>node.get_children()</pre> with you library it does a browse of everything, and then it crashes, because the task is to big for a single browse.

Maybe if i have seen a lower level example of how to retrieve data from OPC.UA, i would have achieved the task with this library.

It is also true that my datasource has to many useless nodes, and we are trying to tell the admins of that datasource to drop them in some way. Nevertheless, it is possible to reach the data :)

Thanks for your help guys. Better than paid support!

starturtle, 2021-01-21:
Wouldn't it help with such issues if `node.get_references` provided an interface to set `RequestedMaxReferencesPerNode` on the [browse parameters](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/node.py#L368)? That may enable a client on slow hardware to run against a server designed for the greedy. Or am I confusing arguments here?

oroulet, 2021-01-21:
Looking at code of get_chilfren() and using the low level api you can do whatever you want. It would have been an easy solution.

But i am interested to know how that parsing error happened. There is s bug somewhere

juamar, 2021-01-21:
> Wouldn't it help with such issues if `node.get_references` provided an interface to set `RequestedMaxReferencesPerNode` on the [browse [...]

Very Greedy like mine. Later last month I knew that the OPC UA Server was done with some C or C++ code kind of fast and quick, by someone with not much time on the server.... neither in his priorities XD

juamar, 2021-01-21:
So cool the commits. Will be nice to test its functionality

AndreasHeine, 2021-01-21:
adding MaxReferencesPerNode to the api makes sens if a client checks the servers operational limits up front! but it dont solve the problem if you have more then +10k subnodes it only works till you reach maxmessagesize then its game over... if a server/client supports a gigantic messagesize  you need to bump up timeouts
tested it with 100k Children (client.uaclient._timeout = 10)
[image]

zerox1212, 2021-01-22:
> but it looks like we never check if a reponse with more then negotiated messagesizes should ever been send out

Should probably at lease be a `log` warning here...