FreeOpcUa/python-opcua #1182: KeyError: 'Node'

FloBer431, 2020-12-04:
Hey Guys
I'm new in programming on phyton and am currently setting up an OPCUA Server on a raspberry pi 4.
The Server should send Temperature und Humidity data.
But when starting the server following Error code is shown:

Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/binary_server_asyncio.py", line 75, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/uaprocessor.py", line 93, in process
    return self.process_message(msg.SequenceHeader(), msg.body())
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/uaprocessor.py", line 113, in process_message
    return self._process_message(typeid, requesthdr, seqhdr, body)
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/uaprocessor.py", line 192, in _process_message
    self.send_response(requesthdr.RequestHandle, seqhdr, response)
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/uaprocessor.py", line 47, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 256, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 296, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 296, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 194, in pack_uatype
    return struct_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 192, in pack_uatype
    return variant_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 378, in variant_to_binary
    b.append(pack_uatype(var.VariantType, var.Value))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 188, in pack_uatype
    return extensionobject_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 458, in extensionobject_to_binary
    TypeId = ua.extension_object_ids[obj.__class__.__name__]
KeyError: 'Node'
Exception in callback InternalSubscription._sub_loop()
handle: <TimerHandle when=12091.963434291 InternalSubscription._sub_loop()>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/internal_subscription.py", line 293, in _sub_loop
    self.publish_results()
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/internal_subscription.py", line 321, in publish_results
    self.callback(result)
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/uaprocessor.py", line 80, in forward_publish_response
    self.send_response(requestdata.requesthdr.RequestHandle, requestdata.seqhdr, response)
  File "/usr/local/lib/python3.7/dist-packages/opcua/server/uaprocessor.py", line 47, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 284, in to_binary
    return struct_to_binary(val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 284, in to_binary
    return struct_to_binary(val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 256, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 296, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 296, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 188, in pack_uatype
    return extensionobject_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 460, in extensionobject_to_binary
    Body = struct_to_binary(obj)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 256, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 296, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 296, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 284, in to_binary
    return struct_to_binary(val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 194, in pack_uatype
    return struct_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 261, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 274, in to_binary
    return pack_uatype(vtype, val)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 192, in pack_uatype
    return variant_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 378, in variant_to_binary
    b.append(pack_uatype(var.VariantType, var.Value))
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 188, in pack_uatype
    return extensionobject_to_binary(value)
  File "/usr/local/lib/python3.7/dist-packages/opcua/ua/ua_binary.py", line 458, in extensionobject_to_binary
    TypeId = ua.extension_object_ids[obj.__class__.__name__]
KeyError: 'Node'

This is my code:

from opcua import Server
from random import randint
import datetime
import time
import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn
from adafruit_ads1x15.ads1x15 import Mode
i2c = busio.I2C(board.SCL, board.SDA)

ads = ADS.ADS1115(i2c)
ads.gain = 1
chan0 = AnalogIn(ads, ADS.P0)
chan1 = AnalogIn(ads, ADS.P1)

server = Server()
url = "opc.tcp://169.254.250.187:4840"
server.set_endpoint(url)
name = "Tempertur-Server-Pi"
addspace = server.register_namespace(name)
node = server.get_objects_node()

Param = node.add_object(addspace,"Parameters")
Temp = Param.add_variable(addspace, "Temperature", 0)
Humi = Param.add_variable(addspace, "Humidity", 0)
Time = Param.add_variable(addspace, "Time", 0)

Temp.set_writable()
Humi.set_writable()
Time.set_writable()

server.start()
print("Server started at {}".format(url))

while True:

        Temperature =((100/32752)*chan0.value)-20
        Humidity =(100/32752)*chan1.value
        TIME = datetime.datetime.now()

        print('{:>5},{:>5}'.format(Temperature, chan0.value))
        print('{:>5},{:>5}'.format(Humidity, chan1.value))

        Temp.set_value(Temperature)
        Humi.set_value(Humidity)
        Time.set_value(Time)

        time.sleep(60)

Maybe some of you can help me.
Thanks

swamper123, 2020-12-04:
I can't point to something directly, but maybe it is caused by your Node definition. The Datatypes may not be clear enough, since you just used a 0, but you have a float/double value.

Try to replace them like this and let us know if it may helped:

```
Temp = Param.add_variable(addspace, "Temperature", 0, ua.VariantType.Float)
Humi = Param.add_variable(addspace, "Humidity", 0, ua.VariantType.Float)
Time = Param.add_variable(addspace, "Time", 0, ua.VariantType.DateTime)

```

FloBer431, 2020-12-04:
Thank you, it worked

closed 2020-12-04 by FloBer431