FreeOpcUa/python-opcua #1183: Regarding Time Out Error In Python OPCUA

rajnishnationfirst, 2020-12-04:
Hi ,

I am using free OPCUA Python for getting values from WinCC Siemens SCADA.

I am facing TIME OUT ERROR issue in 2 cases .

**Case 1) During calling get_value() on  multiple nodes  in a  infinite **while (True) loop**  Like .**

**Code : ( The below code runs continuously and don't have any sleep interval in the while loop  )**

while True :
  ......**.node1.get_value()**

   .....Some processing 

   .....**node2.get_value()**

  .....Some processing 

   .....**node2.get_value()**

The above loop runs continously.

**Case 2) I am calling 22 threads using a THREAD pool which runs parallely and one thread has nearly 150 Tags ( Nodes ) , so 22 thread will have nearly 3300 Tags in total.**

    a) Here EACH Thread in the 22 ThreadPool connects to WinCC Siemens SCADA and using each THREAD using 
     node.read_raw_history(start_time , end_time , 0 ) for each of 150 Tags ...

    b) All 22 threads run parallely uses 3300 Nodes and node.read_raw_history(start_time , end_time , 0 ) to be called all all 3300 
    Tags . every minute .

Also i have set 3600000 Milliseconds  as the timeout in OPCUA client.py Python file  as shown below .
[image]

Also at my WinCC SCADA it has 300000 milliseconds in the WINCC Siemens SCADA configuration file .

What are possible ways to avoid this kind of problem .
 1) Do i need to also set the TIMEOUT at back End WinCC side also as 3600000 Milliseconds as same as in the CLIENT.py file ( Now at WInCC SCADA its 300000 milliseconds)

 2) Do i need to change the python version ?

 3) Also how to restore the connection if a TIMEOUT error / exception occurs and again the program should automatically connect the WinCC SCADA and start fetching the DATA continuously as normal. ?

Pls help me with both the cases that i have raised in the above 2 points .

Thanks 
Rajnish Vishwakarma.

AndreasHeine, 2020-12-04:
polling data in an infinite while loop with node class "get_value" is the worst possible constellation and also with 22 threads 3300 nodes - you flood the network with requests (i would bet 90% of the values does not change from loop to loop) and propably you overwhelm the opc ua server with garbage requests... result timeout!

if you want to read lots of date you should use a subscription or the client class "get_values" method where you can read lots of nodes at once (but there is a messagesize/buffersize limit)
https://github.com/FreeOpcUa/python-opcua/blob/131ad8ceffeb7510b7bb3d9db1742896dbf74196/opcua/client/client.py#L658

get_value -> one request single node (in your case 3300 requests as fast as your pc can throw them...)
get_values -> one request multiple nodes at once
subscription -> you will get notified for each value change of a node

it is also possible that you run into a raise condtion by your own threading code...

vsagarmb, 2021-07-14:
Hello @AndreasHeine 

I am trying to read values from a Kepware UA server and I get the following error. 

Traceback (most recent call last):
  File "opc-client.py", line 106, in <module>
  File "opc-client.py", line 41, in client_run_v1
  File "opcua\common\node.py", line 155, in get_value
  File "opcua\common\node.py", line 164, in get_data_value
  File "opcua\common\node.py", line 275, in get_attribute
  File "opcua\client\ua_client.py", line 347, in read
  File "opcua\client\ua_client.py", line 83, in send_request
  File "concurrent\futures\_base.py", line 447, in result
concurrent.futures._base.TimeoutError

I have tried both get_value() and the get_values() methos and get the same exception. Do you have any pointers for me on how this can be resolved. 

Thanks so much in advance. 

-Sagar

AndreasHeine, 2021-07-14:
@vsagarmb it would be better to open a own issue and fill out the issuetemplate, the error itself can have various reasons...