FreeOpcUa/python-opcua #1185: Using def datachange_notification(self, node, val, data): of Handler

rajnishnationfirst, 2020-12-05:
Hi ,

I need to use def datachange_notification(self, node, val, data): method of the handler class.

a)
I am subscribing a RUNNING TAG1  of a machine to get the value from 1 to 0 .. here 1 indicates machine is running and 0 its stopped .

b)
Here when running bit is 0 , i will check for another TAG2  called as DOWNTIME started ( here i will  note the start time of downtime) , after getting the status of this as 1 , then again i will check for Reason for downtime  TAG3 which has values ranging from 1., 2, 3, 4,.......7 ( which has reasons ) .

c) After i get the reason for downtime TAG3 value , i will check for when the RUNNING TAG1 as 1 , then i will note this end time and calculate Downtime in hours  

The above both point A ,B and C happens when machine is stopped to calculate the downtime of the machine based on RUNNING TAG1 , Downtime Started TAG2 and Reason For Downtime TAG3 .

Consider **TAG 1 - Running BIT , TAG2 - Downtime Started Bit and TAG3 as Reason specified bit**.

@oroulet sir , your suggested to use client.get_values() and subscription logic, but i am not able to apply for the above case.

I tried using infinite loop like 

a) **Using Infinite While Loop:** ( Here below i will check all values instantaneously in infinite while loop based on TAG1 , TAFG2 and TAG3 value.

       ```
 while True:
            global st_time_exe
            print("RUNNING BIT VALUE",self.run_tag.get_value())
            #print("DT STARTED BIT VALUE", self.dt_started.get_value())
            #print("MAIN REASON BIT VALUE", self.main_reason.get_value())
            while True:
                if self.run_tag.get_value()  == True:
                    print("RUNNING BIT STATES RUNNING")
                    break;
                else:
                    print("STOPPED")
                    while True:
                        if self.dt_started.get_value() == True:
                            print("DT STARTED")
                            st_time_exe=time.time();
                            if self.run_tag.get_value() == True:
                                print("RUNNING STARTED IN DT STARTED")
                                break;
                            while True:
                                if self.dt_not_decided.get_value() == False:
                                    print("REASON NOT SPECIFIED")
                                else:
                                    print("REASON SPECIFIED")
                                    if self.run_tag.get_value() == True:
                                        et_time_exe = time.time();
                                        main_reason_specified=self.main_reason.get_value()
                                        array_main_reasons=[]
                                        #postgres_select_query = "SELECT * FROM main_reason_rcbl_stck_erp where value='1'"
                                        #cursor.execute(postgres_select_query)

                                        postgres_select_query = "SELECT * FROM main_reason_rcbl_stck_erp where value = %s"
                                        cursor.execute(postgres_select_query, (str(main_reason_specified),))
                                        tup = cursor.fetchall()
                                        downtime_main_reasons = list(sum(tup, ()))
                                        print(downtime_main_reasons)
                                        print(len(downtime_main_reasons))
                                        main_reason_id = downtime_main_reasons[0]
                                        main_reason_code = downtime_main_reasons[1]
                                        main_reason_type = downtime_main_reasons[2]
                                        main_reason_tagname = downtime_main_reasons[3]
                                        main_reason_tag_node_to_check_subreason = downtime_main_reasons[4]

                                        if main_reason_type=='' or main_reason_type=='':
                                            print("PM & Organisational BLOCK")

                                            array_main_reasons.append("Machine_Name")
                                            array_main_reasons.append(
[... 179 lines cut ...]
        dt_started = client.get_node('ns=1;s=t|DN_STRTED_STCK')
        main_reason = client.get_node('ns=1;s=t|DN_REASONS_NOT_DECIDED_STCK')
        dt_not_decided = client.get_node('ns=1;s=t|MAIN_REASON_STCK')

        sub_reason_tag_code = client.get_node('ns=1;s=t|ELECTRICAL_ERP_STCK')
        print("SUB REASON")
        vr = sub_reason_tag_code.get_value()
        print("SUB2 REASON",vr)

        rcblDowntimeHandlerObj = machine_centre_downtime(client, run_tag,dt_started,main_reason,dt_not_decided)
        time.sleep(1)
    finally:
        #client.disconnect()
        print("myvar is:")
```

In the above both the cases i am facing TIME OUT issue  and how can i do this .

The above is the case for only machine , there will be such 22 machines and i will create muliple objects of HANDLERs .

How to tacke this.

Thanks 

Rajnish Vishwakarma

starturtle, 2020-12-05:
What stops you from deferred execution of your "stop handling routine" on a different thread than your subscription handler? That would immediately unblock the communication thread, and prevent you from needing active polling. And save a ton of network load.

rajnishnationfirst, 2020-12-05:
Thanks @starturtle  for your reply .

Could you pls help me with the an example explaining your statements - **" What stops you from deferred execution of your "stop handling routine" on a different thread than your subscription handler? That would immediately unblock the communication thread, and prevent you from needing active polling. "**

Requesting @starturtle  to pls help me with your guidance that would really help me to solve the problems very fast.

Thanks 

Rajnish Vishwakarma
DI CS SIEMENS INDIA

starturtle, 2020-12-05:
I don't have a dev env for python at hand anytime soon, sorry. But will elaborate, hoping to get you on track.

The core idea would be to create a worker thread waiting for new items in an event queue, and feeding that queue from inside the subscription handler (only). For an intro, see [this tutorial](https://realpython.com/intro-to-python-threading/), for example.

The smallest "queue" is a "conditional variable" (actual or semantical) that waits for the subscription handler to set it. If you have `concurrent.futures` available, it should help a lot. The subscription object needs to stay alive anyway, so you can make arrangements that your objects can reach one another through it.

zerox1212, 2020-12-05:
@rajnishnationfirst you are really asking a lot of open source developers working for free... Questions like this have nothing to do with issues related to the library and are almost entirely about basic python skills. I hope in the future you consider contributing back to the open source projects you are using to make your application.

rajnishnationfirst, 2020-12-06:
> @rajnishnationfirst you are really asking a lot of open source developers working for free... Questions like this have nothing to do with issues related to the library and are almost entirely about [...]

@zerox1212 yes sure .. i will share the hurdles which i faced and facing during the phase of the development.