FreeOpcUa/python-opcua #1191: When subscripting, if opc server disconnect can make error -- "Exception in thread Thread-2". I don't want change souce code but need to inherit many class.

CodePythonFollow, 2020-12-10:
This is a bug report. to get help go to https://github.com/FreeOpcUa/python-opcua/discussions

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior incl code:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

swamper123, 2020-12-10:
Hi,

please provide the informations, which are mentioned in your post. Don't write everythin in the title.

CodePythonFollow, 2020-12-10:
[image]

CodePythonFollow, 2020-12-10:
[image]
main thread can't get the error

swamper123, 2020-12-10:
Catching Exceptions, caused by Threads, in Python is a pain (since ever). You can't pass them to the main thread.
You have to integrate signals or deal with the Exceptions in the Threads themselve or at least join the thread anywhere.

CodePythonFollow, 2020-12-10:
Sure, thank you.Global variable as a integrate signals is good idea.

closed 2020-12-10 by swamper123