FreeOpcUa/python-opcua #1192: Servertimestamp not set automatically of client does not set the Servertimestamp

mbay-SAG, 2020-12-10:
This is a bug report. to get help go to https://github.com/FreeOpcUa/python-opcua/discussions

**Describe the bug**
If you use an OPCUA Client such as UA Expert from Prosys and change values within the gui than the Servertimestamp is not set while writing the node, which is from our perspective not as specified by the OPCUA guidelines.
The Timestamp will be set to EPOCH Time which is on Windows in the year 1601. 
The bug is contained in address_space.py in the write function. There the datavalue for Servertimestamp should be set before writing the node.

**To Reproduce**
Start a simple server with any structure. Use UAEpert, write variables from the GUI. See that Timestamp is set to EPOCH time.

**Expected behavior**
The Servertimestamp should set the timestamp while wirting the node if a new value is set that does not contain the set Servertimestamp.
If the Sourcetimestamp is not set by the client, it is to be discussed whether the SourceTimestamp should be set as well

**Screenshots**
If applicable, add screenshots to help explain your problem.

mbay-SAG, 2020-12-10:
We fixed that in the write function and referenced that already in: 
[https://github.com/FreeOpcUa/python-opcua/issues/850]
If you open a branch we can commit the changed for a pull-request.

AndreasHeine, 2020-12-10:
you can fork -> create a branch -> commit your changes -> PR

mbay-SAG, 2020-12-11:
Thanks @AndreasHeine for referencing to the specs. Just created the PR.

closed 2020-12-11 by mbay-SAG