FreeOpcUa/python-opcua #1200: Subscribe data change

CodePythonFollow, 2020-12-23:
When the subscription data changes, the opc service is disconnected. Is there a corresponding callback function for processing.

AndreasHeine, 2020-12-23:
client or serverside?

CodePythonFollow, 2020-12-24:
client

AndreasHeine, 2020-12-24:
there is no callback for a connection break!

CodePythonFollow, 2020-12-25:
When my subscription data changes, it may be timeout due to a long time. At this time, I have to restart the subscription, so dirty data will be generated and data may be lost.

AndreasHeine, 2020-12-25:
https://reference.opcfoundation.org/v104/Core/docs/Part4/6.7/

just as hint how you can implement your client reconnect!
https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/9f664cd2d6d7f3225ca7334f049c2ff5069ad755/client.py#L62

CodePythonFollow, 2020-12-30:
Thank you. I used the same idea, asynchrony is a great optimization direction