FreeOpcUa/python-opcua #1205: Parsing QualifiedName and NodeId in single from_string(string) function

ArnoLWeiss, 2021-01-04:
**Describe the bug**
The QualifiedName.from_string(string) [function ](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/ua/uatypes.py#L468) attempts to distinguish between strings with and without idx by searching for a ":".

**To Reproduce**
Steps to reproduce the behavior incl code:
1. Create any nodeset-xml that has a BrowseName in URL-Format (including the colon)
2. Load into opcua-modeler

**Expected behavior**
Parse entire BrowseName as string

**Screenshots**
[image]

starturtle, 2021-01-04:
Any browse name is defined to belong to a name space. What would you expect from a browse name parser if no namespace index is given?

ArnoLWeiss, 2021-01-04:
Good point, the possibility to parse a Browsename that brings its own namespace index must be considered. However, this cannot be done by simply splitting at the first ":" as it disregards the possibility that there is NO namespace index given and the name contains a ":" as in "http://opcfoundation.org/UA/Robotics"

oroulet, 2021-01-04:
a browse name must have an index as far as I understand. Does the spec says somewhere you can represent a browse name as a string withotut namespace?

starturtle, 2021-01-04:
A BrowseName is defined as an Attribute of type QualifiedName, which consists of a name string and a namespace index. So no, it is not possible that there simply is "no namespace given". 
If you don't provide a namespace, then the BrowseName is of the wrong data type, unless a default namespace for all nodes can and should be assumed. In that case, it is necessary to agree on one default namespace.

ArnoLWeiss, 2021-01-04:
Sorry for being unclear: I'm not saying that there should be no parsing into namespace index and Browsename. What I am saying is that [this implementation](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/ua/uatypes.py#L471) doesn't consider the case that the Browsename itself might contain a colon (":") and therefore splits where it shouldn't

starturtle, 2021-01-04:
Actually, it does if the full QualifiedName is given as a string. The function will only split the string on the first colon. That's what the second argument is for. What I'm more curious about upon seeing this is whether [escape characters](https://reference.opcfoundation.org/v104/Core/docs/Part4/A.2/) are being respected? I remember being able to TranslateBrowsePathToNodeId for nodes with special characters in their BrowseNames client-side, but I've no experience with the server implementation.

ArnoLWeiss, 2021-01-04:
You're right, the mistake is with me. Thanks for yall's comments. I'll close this.
@starturtle  feel free to tag me if you open an issue or discussion on the escape-char question.

closed 2021-01-04 by ArnoLWeiss