FreeOpcUa/python-opcua #1213: How to read variable from list

TomasTrnkaPLC, 2021-01-07:
Hi.
I not sure why I cannot read data like this
`
OPC_input = PLC_Heart_Beat
value = '"0:Objects", "4:new_Controller_0", "3:GlobalVars" , "4:'
         value_2 = '"'
         tag_string = value + OPC_input + value_2
         print (tag_string)
       
         #OPC_Input_reads = root.get_child(["0:Objects", "4:new_Controller_0", "3:GlobalVars", "4:PLC_Heart_Beat"])
         try:
            OPC_Input_reads = root.get_child([tag_string])
            OPC_Input_reads = OPC_Input_reads.get_value()  
             #globals().update(OPC_input = OPC_Input_read)
            print("OK")

         except:
             print("Cannot be loaded") 
             time.sleep(1)   
          ` but when I use a `OPC_Input_reads = root.get_child(["0:Objects", "4:new_Controller_0", "3:GlobalVars", "4:PLC_Heart_Beat"])` the I can read value.

TomasTrnkaPLC, 2021-01-07:
`  try:
        client.connect()
        client.load_type_definitions()  # load definition of server specific structures/extension objects
        # Client has a few methods to get proxy to UA nodes that should always be in address space such as Root or Objects
        root = client.get_root_node()
        #objects = client.get_objects_node()
               
        for OPC_input in OPC_tag_list_input_array:
         print('First read')
         print (OPC_input)
         value = '"0:Objects", "4:new_Controller_0", "3:GlobalVars" , "4:'
         value_2 = '"'
         tag_string = value + OPC_input + value_2
         print (tag_string)
       
         #OPC_Input_reads = root.get_child(["0:Objects", "4:new_Controller_0", "3:GlobalVars", "4:PLC_Heart_Beat"])
         try:
            OPC_Input_reads = root.get_child([tag_string])
            OPC_Input_reads = OPC_Input_reads.get_value()  
             #globals().update(OPC_input = OPC_Input_read)
            print("OK")

         except:
             print("Cannot be loaded") 
             time.sleep(1)   
          
        client.disconnect()

    except:
         print('Connection Error')  
         
         time.sleep(1)`

oroulet, 2021-01-07:
```
     try:
        OPC_Input_reads = root.get_child([tag_string])
        OPC_Input_reads = OPC_Input_reads.get_value()  
         #globals().update(OPC_input = OPC_Input_read)
        print("OK")

     except:
```
do not reuse the same variable for different things, do not use global(), do ask people to read code full of commented lines...

then maybe someone will spend time helping you

TomasTrnkaPLC, 2021-01-07:
that globals is a scrap from some test. By the way even I rename  OPC_Input_reads = root.get_child([tag_string])
            OPC_Input_reads_variable = OPC_Input_reads.get_value() doesn't help.

TomasTrnkaPLC, 2021-01-07:
Ok. I update a code and now I can read. 
but I cannot write. when I use this.
` try:
             # Check if variable is string  
            if SQL_Variable[0] == '1' or SQL_Variable[0] == '0':
              SQL_Variable = int(SQL_Variable)
              OPC_Output_write = root.get_child(["0:Objects", "4:new_Controller_0", "3:GlobalVars", tag_string])
              Variable = SQL_Variable
              OPC_Output_write.set_attribute(ua.AttributeIds.Value, ua.DataValue(Variable))
            else: 
                print("STRING")
                OPC_Output_write = root.get_child(["0:Objects", "4:new_Controller_0", "3:GlobalVars", tag_string])
                String_var = ua.DataValue(ua.Variant(SQL_Variable, ua.VariantType.String))
                OPC_Output_write.set_value(String_var)  
 
 
         except:
             print("Output "+ OPC_output + " Value:"+ SQL_Variable+" Cannot be set via OPC") 
             time.sleep(1)   
 
        client.disconnect()          
    except:
         print('Output Connection Error') ` why that's finish with a error? Is there a way how to catch exception from OPC?

AndreasHeine, 2021-01-07:
S7-1500:

from opcua import Client, ua

url = "opc.tcp://192.168.10.1:4840"
client = Client(url)
client.session_timeout = 30000
client.connect()
print(f"Connected: {url}")

node = client.get_node('ns=3;s="TEST"."TestUint"')

print("before: ", node.get_value())
node.set_value(ua.DataValue(ua.Variant(5, ua.VariantType.UInt16)))
print("after: ", node.get_value())

client.disconnect()
print("Disconnected")

Connected: opc.tcp://192.168.10.1:4840
before:  200
after:  5
Disconnected

oroulet, 2021-01-08:
closing this since this is not a bug but a discussion

also @TomasTrnkaPLC the problem is that you send us completely unreadbale code to debug (and at the wrong place)

closed 2021-01-08 by oroulet