FreeOpcUa/python-opcua #1240: ArrayDimension is not reported correctly if ValueRank <= 0

DF1AMB, 2021-01-26:
I have used the server-minimal.py example and tried to connect from an Beckhoff TwinCAT OPC UA Client to the python server. That failed with an client error InvalidNodeHdl. After talking to the Technical support of Beckhoff they told me that refering to the spec of OPC UA [1.04](https://reference.opcfoundation.org/v104/Core/docs/Part5/5.4/) the ArrayDimension of the Node has to be NULL or not to be reported. But the OPC UA server reports it as [].

**To Reproduce**<br />
* Start server-minimal.py
* Start a TwinCAT Client
* TwinCAT Client reports Error 0x E4DD 0116 (InvalidNodeHdl)

**Expected behavior**<br />
* TwinCAT Client can connect to the Servernoded Correctly

**Screenshot TwinCAT client -> Python server**<br />
[image]

**Screenshot TwinCAT client -> Codesys server**<br />
[image]

**Version**<br />
Python-Version: 3.8<br />
python-opcua Version: 0.98.12

If there is any information missing I will provide it as fast as possible.

Thanks

oroulet, 2021-01-26:
this is probably the default value of a variable node. I thinkwe fixed that in asyncua.. can you try it? just curious.

DF1AMB, 2021-01-26:
I gave it a try again with the server-minimal.py

[image]

The result is still the same.

And you are also right, this is the default value of a variable node. I tried to change the default to None. When doing this _validate in uatypes.py throws an exception.
`raise UaError("Non array Variant of type {0} cannot have value None".format(self.VariantType))`

oroulet, 2021-01-26:
https://github.com/FreeOpcUa/opcua-asyncio/pull/427
try that one. it should fix it

DF1AMB, 2021-01-26:
Looks promising ...
Will also you provide the solution for this project?

oroulet, 2021-01-26:
I will not have time, but code is very similar so you or someone else can port.
Or you can migrate to asyncua. The sync api seems to work fine

DF1AMB, 2021-01-28:
@oroulet: Thanks for the fix. it works for me now. I have migrated the project to asyncua.

I will also try to port the code to the sync api, since for other projects I am running migration will no be so easy.

lortolan79, 2022-10-14:
Hello.

I'm using asyncua 0.9.98 but I have the same issue: the ArrayDimensions is set to UInt32

[image]

I have checked that this piece of code:

[image]

in "_create_variable" function is performed but the result is not correct.

The variable is created in this way:

[image]

How can I fix it?

Thanks in advance

schroeder-, 2022-10-14:
UInt32 -1 in UaExpert means NULL or in Python None so everything is correct.

lortolan79, 2022-10-14:
Dear schroeder

Are you sure about this?

UAExpert shows two different values depending on which library I use: for example using prosys library or a B&R server the value shown is Null:

[image]

In fact the Beckoff client works fine with B&R or Prosys server while it gives an error (only in writing, reading is correct) with asyncua server.

Thanks for your reply

AndreasHeine, 2022-10-14:
see table 1 -> https://reference.opcfoundation.org/v104/Core/docs/Part5/3.3/

AndreasHeine, 2022-10-14:
https://reference.opcfoundation.org/v104/Core/docs/Part3/5.6.2/

<html><body>

ArrayDimensions | O | UInt32[] | This Attribute specifies the maximum supported length of each dimension. If the maximum is unknown the value shall be 0.The number of elements shall be equal to the value of the ValueRank Attribute. This Attribute shall be null if ValueRank ≤ 0.
-- | -- | -- | --

</body>
</html>

This Attribute shall be **null** if ValueRank ≤ 0

schroeder-, 2022-10-17:
If you want to give the same response as prosys you must set the value of the node to NULL if your array is empty. Which is a uncommon behavior for a server to do and I have never seen from any other opc ua server.

lortolan79, 2022-10-17:
I have not an array...I have a variable with scalar value that has an own -not null- value.
The issue with the Beckhoff client is related to the fact that the value of the attribute "ArrayDimensions" should be "Null". In my server the value of this attribute is set as "None" but in UAExpert is displayed as "UInt32 Array[-1]".
For "non Beckhoff" clients this is absolutely not a problem but for Beckhoff this causes that the variable is not writable.
I'm trying to understand if it is possible to solve this issue in the server side.