FreeOpcUa/python-opcua #1265: Server doesn't handle empty BrowsePath

Pascal-So, 2021-02-19:
**Describe the bug** <br />
The `server-methods.py` example encounters an out of range error.
```
Traceback (most recent call last):
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/binary_server_asyncio.py", line 75, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/uaprocessor.py", line 93, in process
    return self.process_message(msg.SequenceHeader(), msg.body())
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/uaprocessor.py", line 113, in process_message
    return self._process_message(typeid, requesthdr, seqhdr, body)
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/uaprocessor.py", line 269, in _process_message
    paths = self.session.translate_browsepaths_to_nodeids(params.BrowsePaths)
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/internal_server.py", line 369, in translate_browsepaths_to_nodeids
    return self.iserver.view_service.translate_browsepaths_to_nodeids(params)
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/address_space.py", line 147, in translate_browsepaths_to_nodeids
    results.append(self._translate_browsepath_to_nodeid(path))
  File "/home/gea/.local/lib/python3.8/site-packages/opcua/server/address_space.py", line 153, in _translate_browsepath_to_nodeid
    if not path.RelativePath.Elements[-1].TargetName:
IndexError: list index out of range
```

OPC Router then complains about a timeout error because it didn't get a response to the TranslateBrowsePathsToNodeIdsRequest.

**To Reproduce**<br />

Run the `server-methods.py` example and access the server using "OPC Router Version 4.18 - Trial version". The client then eventually sends a "TranslateBrowsePathsToNodeIdsRequest" with an empty array of RelativePathElement (according to wireshark), which causes the error to be printed to the server console.

**Expected behavior**<br />
I'm not sure if this empty path is allowed by the spec, and [a comment in the server source also mentions this](https://github.com/FreeOpcUa/python-opcua/blob/a7f21f738d805c25fa25241e94225ba35b337dab/opcua/server/address_space.py#L155), but apparently this appears in the real world, whether or not it is valid by spec, so I guess this should be handled.

**Version**<br />
Python-Version: 3.8.5
python-opcua Version: 0.98.12
OPC Router Version 4.18 - Trial version

dagsonpatrick, 2024-04-01:
Hello @Pascal-So , I have the same problem as you. Were you able to resolve this issue?

Pascal-So, 2024-04-06:
Hi @dagsonpatrick, unfortunately I wasn't able to fix this issue. If I remember correctly, this was around the time when I was switching everything over to rust anyway, and I've had good experiences with the locka99/opcua library in rust in different projects ever since.