FreeOpcUa/python-opcua #1271: client.load_type_definitions() method got SyntaxError

tom-schulte-wpk, 2021-02-23:
Hi everyone,

i got a syntax error for the load_type_definitions() function by using it as client.

My code is:

`import sys
sys.path.insert(0, "..")
import time
import logging
from IPython import embed

from opcua import Client
from opcua import ua

logging.basicConfig(level=logging.WARN)

client = Client("opc.tcp://192.167.10.1:4840")

try:
    client.connect()
    
    # Client has a few methods to get proxy to UA nodes that should always be in address space such as Root or Objects
    root = client.get_root_node()
    print("Objects node is: ", root)
    
    # Node objects have methods to read and write node attributes as well as browse or populate address space
    print("Children of root are: ", root.get_children())
    
    node_1 = client.get_node("ns=3;s=PLC")
    node_2 = client.get_node('ns=3;s="DatabaseCom_DB"')
    
    objects = client.get_objects_node()
    print(objects)
      
    
    node_test = client.get_node(('ns=3;s="DatabaseCom_DB"."ctrl".'))
    print(node_test)
    
    
    print(node_1.get_browse_name())
    print(node_2.get_browse_name())
    
    node_1_vars = node_1.get_variables()
    node_2_vars = node_2.get_variables()
    print(node_2_vars)
    var_test = node_2_vars[0]
    print(var_test)
        
    values_2 = []
    for i in range (len(node_2_vars)):
        print(node_2_vars[i])
        values_2.append(node_2_vars[i].get_value()) 
        
    value_test = values_2[0]
    print(value_test)
    
    client.load_type_definitions()
    
    
finally:
    
    client.disconnect()`

the output and error message is:

**WARNING:opcua.client.client:Requested session timeout to be 3600000ms, got 30000ms instead
Objects node is:  i=84
Children of root are:  [Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=87))]
i=85
ns=3;s="DatabaseCom_DB"."ctrl".
QualifiedName(3:kn-plc-1k1)
QualifiedName(3:DatabaseCom_DB)
[Node(StringNodeId(ns=3;s="DatabaseCom_DB"."ctrl")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."config")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."batchID")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."actMixingTime")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."dateAndTime")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."recipeNumber")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."recipeName")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."reserve")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."cyclicData")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."acyclicDataWrite")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."acyclicDataRead")), Node(StringNodeId(ns=3;s="DatabaseCom_DB"."databaseStatus"))]
ns=3;s="DatabaseCom_DB"."ctrl"
ns=3;s="DatabaseCom_DB"."ctrl"
ns=3;s="DatabaseCom_DB"."config"
ns=3;s="DatabaseCom_DB"."batchID"
ns=3;s="DatabaseCom_DB"."actMixingTime"
ns=3;s="DatabaseCom_DB"."dateAndTime"
ns=3;s="DatabaseCom_DB"."recipeNumber"
ns=3;s="DatabaseCom_DB"."recipeName"
ns=3;s="DatabaseCom_DB"."reserve"
ns=3;s="DatabaseCom_DB"."cyclicData"
ns=3;s="DatabaseCom_DB"."acyclicDataWrite"
ns=3;s="DatabaseCom_DB"."acyclicDataRead"
ns=3;s="DatabaseCom_DB"."databaseStatus"
ExtensionObject(TypeId:StringNodeId(ns=3;s=TE_"2432762033"), Encoding:1, 3 bytes)
  File "<string>", line 47
    self.global = ua.MSA_MStatus_Global_TYP()
         ^
SyntaxError: invalid syntax**

**Version:**
python version is   3.8.5
opcua                     0.98.12                 pypi_0    pypi
opcua-client           0.8.0                     pypi_0    pypi
opcua-modeler      0.5.10                   pypi_0    pypi
opcua-widgets       0.5.10                   pypi_0    pypi
openssl                   1.1.1j                    h2bbff1b_0

I want to change the ExtensionObject with the load_type_definitons() function to be able to read it for further operations. However I got the above mentioned syntax error. Mabe someone has a solution for this issue.

AndreasHeine, 2021-02-23:
client.load_type_definitions()
just adds the type to ua.uaprotocol_auto as a class while runtime!
and while doing that it generates code which crashes...

could you provide more info about the Strucure/CustomType you try to load?
UaExpert AttributesTab of that Structure/CustomType

_just as a hint always load the types after connect!_

tom-schulte-wpk, 2021-02-23:
[image]

 
Is this info ok for you or do you need more info. 

Loading the types after connect you mean like this:
`client = Client("opc.tcp://192.167.10.1:4840")

try:
    client.connect()
    client.load_type_definitions()
`

AndreasHeine, 2021-02-23:
s7 ;)

AndreasHeine, 2021-02-23:
you wrote "I want to change the ExtensionObject with the load_type_definitons() function to be able to read it for further operations"

the ExtensionObject is defined by the Server/PLC and cant be changed by the client!

after loading the typedefinitions you should be able to read them like you would with other variables
```python
val = node.get_value()
field1 = val.field1
field2 = val.field2
```

 if you want to write a variable with an extensionobject as type you need to get an class instance of that exobj:
```python
extobj = ua.uaprotocol_auto.mytype() # mytype is only avalible after load typedefinitions
extobj.field1 = 0
extobj.field2 = "---"
client.write(ua.Variant(val, ua.VariantType.ExtentionObject))
```

maybe this helps a little how to handle extobj (its not with a custom type but it should be similar):
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-extension-object-as-method-argument.py

tom-schulte-wpk, 2021-02-23:
First of all thanks for your fast help. Is this a "typical" problem with Siemens S7 in combination with the struct datatype? Would a server/plc from Beckhoff or B&R work with python-opcua and similar struct datatypes. 

With Siemens you could also try to use node-red to get the values. However my goal was to get the "whole" node from the server and not only read the values of the nodes. Therfore I tried to use the python-opcua.

Maybe I am asking stupid questions, because I am new to this topic. 😅

AndreasHeine, 2021-02-23:
what do you meen with "whole node"?

everything in opc ua is standardized the only difference could be in the opc ua version 1.03 and 1.04 but it depend on the serverimplementation!

tom-schulte-wpk, 2021-02-23:
In my example it would be the node "DatabaseCom_DB" with its nodes. 

You mean the opc ua version in the plc/server (siemens s7)?

oroulet, 2021-02-23:
Most probably some objects have a very strange name that breaks the python syntax. Any ideas which one? We have a cleaning function but it does not catch everything

oroulet, 2021-02-23:
Looking at your output maybe they have quotes?

AndreasHeine, 2021-02-23:
> You mean the opc ua version in the plc/server (siemens s7)?

maybe debend on the firmware (from siemens i know they changed a lot from 2.6 uppwards)

> In my example it would be the node "DatabaseCom_DB" with its nodes.

if you want all subnodes you need to call "node.get_children()" to the "DatabaseCom_DB"-Node (but its not recursive so its not a cralwer)

and with that list of nodes you can do stuff like "client.get_values(nodelist)" if you want their value or "client.uaclient.get_attributes(nodelist, ua.AttributeIds.<AttributeName>)" if you want something else like browsename displayname, ...

https://github.com/FreeOpcUa/python-opcua/blob/0d595e596c9922699f0b36abb79583d5763eeb9b/opcua/ua/attribute_ids.py#L5

>     NodeId = 1
>     NodeClass = 2
>     BrowseName = 3
>     DisplayName = 4
>     Description = 5
>     WriteMask = 6
>     UserWriteMask = 7
>     IsAbstract = 8
>     Symmetric = 9
>     InverseName = 10
>     ContainsNoLoops = 11
>     EventNotifier = 12
>     Value = 13
>     DataType = 14
>     ValueRank = 15
>     ArrayDimensions = 16
>     AccessLevel = 17
>     UserAccessLevel = 18
>     MinimumSamplingInterval = 19
>     Historizing = 20
>     Executable = 21
>     UserExecutable = 22
>     DataTypeDefinition = 23
>     RolePermissions = 24
>     UserRolePermissions = 25
>     AccessRestrictions = 26
>     AccessLevelEx = 27

tom-schulte-wpk, 2021-02-24:
@oroulet 
> Looking at your output maybe they have quotes?

Maybe the plc (s7) created it somehow.

@AndreasHeine 
>maybe debend on the firmware (from siemens i know they changed a lot from 2.6 uppwards)

I will look after the firmware maybe this can solve the problem of the struct datatypes of this nodes.

>if you want all subnodes you need to call "node.get_children()" to the "DatabaseCom_DB"-Node (but its not recursive so its not a cralwer)

>and with that list of nodes you can do stuff like "client.get_values(nodelist)" if you want their value or "client.uaclient.get_attributes(nodelist, ua.AttributeIds.)" if you want something else like browsename displayname, ...

the result is the same as for my example. The struct datatypes of the plc will change to the datatype ExtensionObject. Maybe one more question to the get_attributes function. Should i use the ID-number for the respective attribute e.g. 1 for NodeID like this
**get_attributes(nodeslist,1)**

AndreasHeine, 2021-02-24:
get_attributes(nodeslist, ua.AttributeIds.NodeId) # more readable