FreeOpcUa/python-opcua #1272: Tags are not updating from server side.

jigar88, 2021-02-25:
Hello I am working on OPCUA server side using this python library. I am updating tags based on the messages I received from other service. Server is working for few hours fine but after that tags are not updating even server stays connected. Here are logs. 

Until tags are updating logs looks like:
```
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
INFO: subscription <opcua.server.internal_subscription.MonitoredItemService object at 0x7f8afb25b198>: datachange callback called with handle '202' and value 'Variant(val:2105081,type:VariantType.Int64)'
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: StringNodeId(ns=2;s=/WH/S/401/test_int4) AttributeIds.Value DataValue(Value:Variant(val:8,type:VariantType.String), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:11:56.703182)
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: StringNodeId(ns=2;s=/OPCUA_SERVER/HA/TotalMessagesGenerated) AttributeIds.Value DataValue(Value:Variant(val:2105082,type:VariantType.Int64), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:11:56.703607)
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
INFO: subscription <opcua.server.internal_subscription.MonitoredItemService object at 0x7f8afb25b198>: datachange callback called with handle '201' and value 'Variant(val:2105082,type:VariantType.Int64)'
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: StringNodeId(ns=2;s=/OPCUA_SERVER/HA/TotalMessagesProcessed) AttributeIds.Value DataValue(Value:Variant(val:2105082,type:VariantType.Int64), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:11:56.704350)
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
```

then suddenly it stopped updating tags although server status is connected. In logs I am not seeing INFO for subscription after few hours when tags not updaing . 

Sample Logs when tags not updating are 
```
+++++++++++++++++++++++++++
INFO: subscription <opcua.server.internal_subscription.MonitoredItemService object at 0x7f8afb25b198>: datachange callback called with handle '202' and value 'Variant(val:2105082,type:VariantType.Int64)'
DEBUG: sending datachanges notification with 36 events
INFO: forward publish response PublishResult(SubscriptionId:{self.SubscriptionId}, AvailableSequenceNumbers:{self.AvailableSequenceNumbers}, MoreNotifications:{self.MoreNotifications}, NotificationMessage:{self.NotificationMessage}, Results:{self.Results}, DiagnosticInfos:{self.DiagnosticInfos})
DEBUG: received 112 bytes from socket
INFO: publish request
INFO: publish request with acks [SubscriptionAcknowledgement(SubscriptionId:{self.SubscriptionId}, SequenceNumber:{self.SequenceNumber})]
INFO: publish request with acks [13040]
INFO: publish forward to server
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:11:57.411104,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:11:57.411150)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:11:58.439216,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:11:58.439251)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:11:59.482507,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:11:59.482558)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:00.525465,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:00.525499)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:01.546294,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:01.546327)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:02.600479,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:02.600534)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:03.638818,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:03.638852)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:04.705806,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:04.705841)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:05.726438,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:05.726480)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:06.763102,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:06.763146)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:07.798037,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:07.798072)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:08.817079,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:08.817112)
DEBUG: received 128 bytes from socket
INFO: Read request
DEBUG: read ReadParameters(MaxAge:{self.MaxAge}, TimestampsToReturn:{self.TimestampsToReturn}, NodesToRead:{self.NodesToRead})
DEBUG: get attr val: FourByteNodeId(i=2259) 13
INFO: sending read response
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:09.894750,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:09.894788)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:10.916077,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:10.916116)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:11.940880,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:11.940915)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:12.968417,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:12.968450)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:13.990813,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:13.990843)
[... 46 lines cut ...]
[... cut ...]
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:43.877979,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:43.878009)
DEBUG: write WriteParameters(NodesToWrite:{self.NodesToWrite}) as user User.Admin
DEBUG: set attr val: NumericNodeId(i=2258) AttributeIds.Value DataValue(Value:Variant(val:2021-02-25 19:12:44.919859,type:VariantType.DateTime), StatusCode:StatusCode(Good), SourceTimestamp:2021-02-25 19:12:44.919888)
```

AndreasHeine, 2021-02-26:
please stick to the issue template!
no one here knows anything about your application!

in the logs is no errormessage whatsoever... how should we know what the issue is?

jigar88, 2021-03-03:
Opening new one with template format.

closed 2021-03-03 by jigar88

jigar88, 2021-03-03:
https://github.com/FreeOpcUa/python-opcua/issues/1276